victronenergy/gui-v2 #928: HTML splash screen layout is quiet different to Qt splash screen, causing flicker (wasm)

jpetrell, 2024-04-08:
There are quite many visual differences between the current HTML splash screen and Qt-side implementation. Ideally the layouts would align so the transition would be as smooth as possible to avoid unnecessary flicker and provide a more calm, professional impression.

Currently the HTML splash screen font does not follow the brand font defined by the Victron UI guidelines, the Victron logo position and size is different to the Qt-side splash screen and the labels "Qt for WebAssembly: venus-gui-v2" and "Loading..." are different to Qt-side's initial label "[1] Connecting".

- Align layouts, avoid unnecessary jump in the transition
- Only display minimal labels e.g. just "Loading...". On desktop the loading is so fast that you can hardly even read anything
- Qt framework and WebAssembly are technical details that have no meaning to the end user

jpetrell, 2024-04-10:
Discussed with Serj. Both cases should follow the same design that has already been defined in Figma mock-ups, no talk about Qt or WebAssembly, the layouts aligning as well as possible when transitioning between the HTML and Qt views. "Loading..." label can be shown initially if the progress bar shown on the design is difficult to do with HTML.

mpvader, 2025-06-05:
is going to be fixed soon as part of another issue or PR, closing

mpvader, 2025-06-05:
#2142

closed 2025-06-05 by mpvader