victronenergy/gui-v2 #931: Migrate remaining Bluetooth bits that rely on Connman to Venus platform

jpetrell, 2024-04-10:
Wi-Fi and Ethernet settings are migrated (#618) from Connman to Venus platform APIs to e.g. allow configuring settings remotely with WebAssembly build. There is still one thing that relies on Connman directly, the availability of Bluetooth modem is checked from Connman `Connman.technologyList.indexOf("bluetooth") !== -1`. Discussed with Rein that we could instead make `/Settings/Services/Bluetooth` conditional on the presence of Bluetooth. This would allow removing Connman dependency in both GUI v1 and v2. The Bluetooth availability check is used to show/hide Bluetooth and Bluetooth sensor settings pages.

ReinvdZee, 2024-04-10:
@jhofstee, 

Would you suspect any problems with venus-platform creating `/Settings/Services/Bluetooth` only when the device has bluetooth capabilities? Connman API has been migrated to venus-platform anyway so we can check if "bluetooth" is in the technology list, like the gui currently does.

jhofstee, 2024-05-23:
since Bluetooth support is dynamic, e.g. with an USB module I would prefer not to combine the two, so I added a separate item:

Network/HasBluetoothSupport -> There is either onboard or external Bluetooth support
Settings/Services/Bluetooth -> Start the VE.Smart service for Bluetooth

blammit, 2024-09-24:
Needs to be implemented for 1.0.0.

commit 716313a4 (2024-09-25): Migrate remaining Bluetooth bits that rely on Connman to Venus platform

Fixes #931
commit 55a5f2ab (2024-09-30): Migrate remaining Bluetooth bits that rely on Connman to Venus platform

Fixes #931
PR #1504 (2024-09-30): Migrate remaining Bluetooth bits that rely on Connman to Venus platform
Fixes #931
commit 1ae5d98d (2024-09-30): Migrate remaining Bluetooth bits that rely on Connman to Venus platform

Fixes #931
closed 2024-09-30 by DanielMcInnes