victronenergy/gui-v2 #945: wasm builds: scale is not defined

DanielMcInnes, 2024-04-15:
On wasm builds only, I see the following error:

qrc:/venus-gui-v2/Main.qml:86:3: ReferenceError: scale is not defined

jpetrell, 2024-04-16:
Line 86 points to contentItem bindings. There is one signal handler and one binding that touches the scale property, but yeah seem to work (and actually needed specifically for wasm) and at least from the surface correct:

```
contentItem {
  onScaleChanged: Global.scalingRatio = contentItem.scale
  scale: Math.min(1.0, root.width/Theme.geometry_screen_width, root.height/Theme.geometry_screen_height)
}
```

> The code looks fine, this should be ok, but for some reason it isn't.

Speculating here, but QML is known to sometimes print bogus warnings, it suppresses many usual warnings during initialization before everything is available, but sometimes bogus warnings leak out like how you see parent warnings if you don't write "Item { width: parent ? parent.width : 0 }" even though from QML code point of view the parent should be there from the start. Root item in the scene graph is probably not that often accessed like this.

chriadam, 2024-04-16:
Try changing `Global.scalingRatio = scale` perhaps, also

jpetrell, 2024-04-18:
> Try changing Global.scalingRatio = scale perhaps, also

Tested. Results in "ReferenceError: scale is not defined" since property grouping doesn't change the local scope to contentItem, and the Window has no scale, but contentItem.scale. Though does seem a bit unintuitive that you can refer to Item properties inside Item { .. } block, but not to contentItem properties inside contentItem { .. } block.

chriadam, 2024-04-19:
Ah, I see!

blammit, 2024-11-25:
No longer reproducible. Reopen again if needed.

closed 2024-11-25 by blammit