victronenergy/gui-v2 #946: Add empty state views where applicable

blammit, 2024-04-15:
See Figma. Serj has provided empty state views, with specific examples for:

- Generator start/stop settings, when there is no generator
- Wi-Fi settings, when no adaptor is connected
- Levels page, when there are no tanks or environment sensors
- Overview page, when there are no left-hand boxes (i.e. no AC inputs, DC inputs or PV devices)

As part of this task, investigate other cases where these can be used.

jpetrell, 2024-04-16:
I have been discussing the empty states with Serj as part of #723. There hasn't been empty state design before really, which has resulted different views implementing the empty state in their own way.

jpetrell, 2024-04-16:
Marking as duplicate.

closed 2024-04-16 by jpetrell

blammit, 2024-04-17:
> I have been discussing the empty states with Serj as part of #723. There hasn't been empty state design before really, which has resulted different views implementing the empty state in their own way.

Ah! Thanks :)