victronenergy/gui-v2 #978: Profile and improve GUI v2 launch time and memory footprint

jpetrell, 2024-04-22:
On Ekrano gui-v2 takes 80MB of unique memory after launch and dozen seconds to fully become usable, which seems like a lot for a single-process Qt Quick app (tested with file-system builds). In gui-v2 the dozens of models get constructed on launch and then constantly tracked on real-time, whether the loaded data points are shown on the screen or not. Usually it is good idea to load the models lazily instead, i.e. when the view that requires the model becomes visible.

Overall the current approach is understandable since the Brief and the Overview pages need to know a lot about the connected devices, and because we want to keep the all the main views (Brief, Overview, Levels, Notifications, Settings) in the memory for faster access (i.e. sacrifice bit of launch time for smoother experience during the usage).

Still it would be good to profile a bit what is happening under the hood, what kind of things are slowing down the launch and contributing the overall memory footprint to identify issues and potential places of improvement.

chriadam, 2024-04-22:
I think improving launch speed is mostly important for WebAssembly builds, since the user is likely to launch the application frequently via VRM.

For device builds, I suspect it makes more sense to trade slightly longer launch speed for faster transitions etc (i.e. don't do lazy loading of new models etc) since it's expected that the app will restart very rarely.  Certainly this is the case for the main pages of the app, and probably the drilldowns from the Overview page, perhaps not so much for the Settings pages as you point out - but I'm no expert about how often those are accessed, I'm sure Jesús would have a more informed opinion than mine. In any case, due to the performance impact of memory pressure and real-time data tracking for all subpaths etc, it could very well be that loading lazily is better in all cases, regardless.

In short, I agree that more benchmarking is definitely required, and that there are a lot of improvements which definitely make sense as you point out.

chriadam, 2025-11-06:
Some of this has been done (e.g. migrating models to C++) but there is far more we can do, as Joona suggested.