victronenergy/gui-v2 #984: Explore possibility of automating GUI v2 UI testing

jpetrell, 2024-04-23:
Why
- There are too many features to maintain by a small team
- Manually testing 169 pages after every code change is not practical

Investigate possibility of
- Toggling through all the main views and all the mock configs (26 brief/overview configs, 13 tank/sensor configs, sidePanel.showing true/false, Global.mainView.controlsActive true/false, etc.)
- Iterating through all the available sub-pages pages by recursively finding all ListNavigationItems in the current page
- Running the tests on different languages, color schemes (dark/light), layout configurations (five/seven inch) and system units (electric, temperature, volume, brief levels)
- Generating screenshots of each opened page, and reporting any loading errors and warnings

Automated UI testing can be a huge time sink so need to be (effort vs. issues found) so need to be careful with this. Prioritise testing with real backends, but different layout configurations are probably better tested with the mock backend. Bea has built nice mocking capabilities, which make testing a breeze.

chriadam, 2025-05-30:
Subtasks of this might be:

1) giving unique (e.g. filename-prefixed) object names to every object we need to interact with programmatically, and adding a way to get a pointer to an object with a given name (by walking the child object hierarchy from the root window, for example).

2) creating the mock profile and testing-mode config (i.e. with stable values which aren't being animated, to ensure that the test conditions are stable).  We will actually need multiple such profiles to test different system variants (e.g. 3 phase vs 1 phase AC, EVCS vs no EVCS, critical loads vs none, a variety of different inputs e.g. wind, solar, generator, alternator, grid, shore, etc etc), and a way to launch the app with that profile, e.g. command line arg or whatever.

3) figuring out how/where to store the baseline images (they should probably NOT be part of the main repository, IMO, as there are too many of them).  I suspect creating another git repro like gui-v2-test-baselines or something is the correct solution, but not 100% sure.

4) writing some little application which allows simple comparison of the results would also be useful, e.g. you run the full test and then it will save all of the failed results (along with the baseline) to some temporary folder.  Then you run the "comparison" app to cycle through them one-by-one, and allows you to toggle on/off highlights (in red or whatever) the different pixels between each image, and then choose "update baseline" if the new image is correct (clicking "update baseline" should push that image to the git repo from (3) overwriting the old baseline).  Otherwise, clicking "acknowledge failure" just proceeds to the next failure comparison.

5) then writing the actual tests as separate, standalone, hierarchical test cases, which basically look like: `"click object "foo"; wait transition-duration; click object "bar"; wait transition duration; window.grabImage(); saveResult("test_case_name");`

MikeTrahearn-Qinetic, 2025-06-02:
Is there a reason why we aren't using Squish for this?
The size of the task outlined in these points could amount to a possible diminishing return on cost of hours vs ROI so I would value an objective discussion before committing to reinventing a large number of wheels.
When used, it can use QML ids rather than objectNames (other than e.g. in delegates where you can synthesize one using an index).

chriadam, 2025-06-02:
I think the scope of what we're doing is really small (basically just: click button, take screenshot, compare), and can easily be implemented.  I personally would like to avoid pulling in external dependencies if possible, as every external dependency increases complexity, requires additional learning, and (at least in the case of squish) entails monetary cost.

Effectively, I believe that the UI tests we need to do can be implemented as follows:
```
namespace Victron {
namespace VenusOS {
namespace GuiV2 {

typedef TestInteraction QPair<QString, int>;    // each interaction consists of a globally unique object name (to click), plus a delay (transition delay).
typedef TestStep        QList<TestInteraction>; // each test step consists of a series of clicks+waits through the UI (starting from the brief page view).
typedef TestReset       QList<TestInteraction>; // how to reset back to brief page view after the test step is complete.
typedef TestCase        QPair<TestStep, TestReset>; // each test case consists of a series of steps, image capture+comparison, then state reset steps.
typedef TestSuite       QList<TestCase>;        // each test suite consists of a series of test cases.

class UiTests : public QObject
{
    Q_OBJECT

private:
    QQuickWindow *m_window = nullptr;
    int m_whichCase = 0;
    int m_whichStep = 0;
    int m_whichReset = 0;
    TestSuite m_suite;
    QString m_suiteName;

public:
    explicit UiTests(QObject *parent = nullptr) : QObject(parent) {}

    void start(QQuickWindow *window, const QString &mockConfig) {
        m_suiteName = mockConfig;
        m_window = window;
        if (mockConfig == QStringLiteral("some_config_one")) {
            m_suite = someConfigOne();
        } else if (mockConfig == QStringLiteral("some_config_two")) {
            //m_suite = someConfigTwo();
        } // etc

        next();
    }

    void next() {
        if (m_whichCase >= m_suite.size()) {
            emit finished();
            return;
        }

        const TestCase case = m_suite.at(m_whichCase);
        const TestStep step = case.first;
        const TestReset reset = case.second;

        if (m_whichStep < step.size()) {
            doInteraction(step.at(m_whichStep));
        } else if (m_whichReset < reset.size()) {
            doInteraction(reset.at(m_whichReset));
        } else {
            emit error();
            return;
        }
    }

    void doInteraction(const TestInteraction &interaction) {
        QQuickAbstractButton *button = findButton(interaction.first);
        if (!button) {
            emit error();
            return;
        }
        button->click();
        QTimer::singleShot(interaction.second, this, &UiTests::transition);
    }

    void transition() {
        const TestCase case = m_suite.at(m_whichCase);
        const TestStep step = case.first;
        const TestReset reset = case.second;

        if (m_whichStep < step.size()) {
            m_whichStep++;
            if (m_whichStep >= step.size()) {
                // finished performing test step interactions.
                grabImage();
                QTimer::singleShot(1, this, &UiTests::next);
            }
        } else if (m_whichReset < reset.size()) {
            m_whichReset++;
            if (m_whichReset >= reset.size()) {
                // finished resetting UI state after performing this test case.
                m_whichStep = 0;
                m_whichReset = 0;
                m_whichCase++;
                QTimer::singleShot(1, this, &UiTests::next);
            }
        }
    }

    void grabImage() {
        const QString &baseline_filename = QStringLiteral("%1_%2").arg(m_suiteName).arg(m_whichCase);
        const QString &screenshot_filename = QStringLiteral(%1_%2").arg(baseline_filename, QStringLiteral("result"));
        const QImage baseline(baseline_filename);
        const QImage screenshot = m_window->grabImage();
        if (screenshot != baseline) {
            screenshot.save(screenshot_filename);
            emit failed(m_whichCase);
        }
    }

Q_SIGNALS:
    void error();
    void failed(int whichCase);
    void finished();

private:
    QQuickAbstractButton* findButton(const QString &objectName) {
        const QObject *root = m_window ? m_window->contentItem() : nullptr;
        return root ? root->findChild<QQuickAbstractButton>(objectName) : nullptr;
    }

    TestSuite someConfigOne() {
        TestSuite suite;

        TestStep step_one;
        step_one.append(qMakePair(QStringLiteral("navbar_briefpage"), 1000));
        TestReset reset_one;
        reset_one.append(qMakePair(QStringLiteral("navbar_briefpage"), 1000)); // unnecessary in this case, but for completeness
        TestCase case_one = qMakePair(step_one, reset_one);
        suite.append(case_one);

        TestStep step_two;
        step_two.append(qMakePair(QStringLiteral("navbar_briefpage"), 1000));
        step_two.append(qMakePair(QStringLiteral("briefpage_sidepanel"), 1000));
        TestReset reset_two;
        reset_two.append(qMakePair(QStringLiteral("briefpage_sidepanel"), 1000));
        reset_two.append(qMakePair(QStringLiteral("navbar_briefpage"), 1000)); // unnecessary in this case, but for completeness
        TestCase case_two = qMakePair(step_two, reset_two);
        suite.append(case_two);

        // etc.

        return suite;
    }
};

} // GuiV2
} // VenusOS
} // Victron
```

chriadam, 2025-11-06:
Bea has made significant progress on this, but we will push it back to v1.3.0 in order to focus on polish for v1.2.0 for now.  I have made it high-prio for v1.3.0, and will need strong arguments to convince me otherwise ;-)