victronenergy/gui-v2 #1003: Tank gauges on randomly trail behind the actual remaining capacity

jpetrell, 2024-04-29:
Randomly the fill bar does not appear on a tank or several tanks, note black water at 27%. This has occurred viewing gui2 through VRM and on the Cerbo GX Touch. Screenshots taken a few minutes apart."

[image]
[image]
[image]

chriadam, 2024-05-16:
Can reproduce.  The value in BarGauge.qml goes to NaN for some reason in some cases.

chriadam, 2024-05-16:
... but sometimes even when it isn't NaN, the rectangle doesn't render, even though the "nextPos" value is calculated correctly.  Will investigate further :-/

closed 2024-05-22 by chriadam

jpetrell, 2024-05-22:
Saw this when looping through the tank configurations on the latest main with the mock backend: 
[image]

commit 9ab44faf (2024-05-16): Improve rendering of BarGauge

- Only update the _value if the value is valid (!isNaN())
- Always calculate the _nextPos on _value change, regardless of
  item visibility
- If animating to _nextPos, ensure we stop any current animation
  and set the "from" value appropriate, to avoid "over animating".

Fixes #1003
commit 4db3b392 (2024-05-16): Improve rendering of BarGauge

- Always calculate the _nextPos on _value change, regardless of
  item visibility
- If animating to _nextPos, ensure we stop any current animation
  and set the "from" value appropriate, to avoid "over animating".

Fixes #1003
commit 5ab28324 (2024-05-16): Improve rendering of BarGauge

- Always calculate the _nextPos on _value change, regardless of
  item visibility
- If animating to _nextPos, ensure we stop any current animation
  and set the "from" value appropriate, to avoid "over animating".

Fixes #1003
commit a794af70 (2024-05-17): Improve rendering of BarGauge

- Always calculate the _nextPos on _value change, regardless of
  item visibility
- Only animate to _nextPos if the previous animation has
  completed, to avoid "over animating"
- Enable a layer for the source item
- Specify explicit z ordering for each element

Fixes #1003