victronenergy/gui-v2 #1008: Solar yield widget displays watts even when the user has requested to see amps

jpetrell, 2024-04-29:
Go to Settings > Display & Language > Units. Set Electrical power display preference to amps.

Expected result: Electrical quantities shown on the Brief and Overview pages respect the setting.

Actual result: The solar overview widget still reports values in watts.

[image]

Furthermore, on ESS demo mode (set under General > Demo modes) everything (both the Brief and Overview) is in watts, regardless of what has been specified in the Units settings.

jpetrell, 2024-04-30:
Also reported [here](https://community.victronenergy.com/questions/270759/venus-os-v3401-available-for-testing.html) in the earlier release feedback:
- "Gui-V2: changing measurements unit from W to A (Display>), the load devices (right side of display) change in A, while the charge ones (left side) remain in W"

blammit, 2024-04-30:
Could it be that the system has both AC and DC sources for solar, so then gui-v2 has to show watts here instead of amps? 

https://github.com/victronenergy/gui-v2/pull/1024 updates the unit settings page to clarify this.

blammit, 2024-05-09:
Actually, there are some bugs in how the solar watts/amps are displayed, as per https://github.com/victronenergy/gui-v2/issues/1036. Will close this bug as a duplicate.

closed 2024-05-09 by blammit