victronenergy/gui-v2 #1016: Improve solar yield widget on the Overview page

jpetrell, 2024-04-30:
[Reported](https://community.victronenergy.com/questions/270759/venus-os-v3401-available-for-testing.html) by the customer:
- "I still hope that the solar graph would be much more detailed, scientific as that's what eveyone is eyeballing constantly."
- "Also, am I wrong in saying that the solar graph only shows the history as far as the gui has been running? That's ought to change I presume."

Also [here](https://community.victronenergy.com/questions/279997/venus-os-v3408-available-for-testing.html):
- "Couls you please add: 2. Show more details (Currant, Voltage) at "Overview" for Solar- like at battery"

[image]

blammit, 2025-06-23:
This shows the comparative yield. Depending on the solar device, it may only support two days of history, so that is what you are seeing here.

@philipptrenz FYI, for things to consider in feature updates.

Jalle19, 2025-06-23:
I have an MPPT RS. I guess the problem is that the nights are too bright this time of year so it doesn't understand when a new day begins, thus it's currently on its second day even though the device has been in use for two weeks soon. It defaults to starting a new day at around ~16:00 PM which is when I initially commissioned the unit.

Jalle19, 2025-06-23:
https://community.victronenergy.com/t/mppt-rs-only-shows-two-days-worth-of-history-on-gx-device-full-history-in-victronconnect/36914/2?u=jalle19

This should perhaps be taken into account in the UI design