victronenergy/gui-v2 #1019: Higher precision requested for the battery voltages

jpetrell, 2024-04-30:
[Reported](https://community.victronenergy.com/questions/270759/venus-os-v3401-available-for-testing.html) by the customers: 
- "The Battery details display on the Touch 50 and Cerbo-s GX only shows cell voltages to one decimal place. The current released version shows to 3 decimal places. We find this fine resolution invaluable when a customer is recovering a flat battery: both when the battery is at 0%SOC and being changed and then when it is full charged to determine if any cells are damaged. We typically operate aroun 0.003V difference between low and high. lease can we have this high definition resumed."

[image]

- "In GUI-V2 the min and max battery voltage on the BMS is shown with one digit precision. That is not really useful on LiFePo4 batteries. The old version had 3 digits available. Please update this to 3-digits again.)"
[image]

chriadam, 2024-04-30:
@jepefe raising this for your consideration

jepefe, 2024-05-01:
@chriadam @jpetrell Please, set these voltages to 3 decimals resolution for this product.

jpetrell, 2024-05-03:
> @chriadam @jpetrell Please, set these voltages to 3 decimals resolution for this product.

Ideally would avoid product-specific variations, but ok. I don't know what battery the first user was using, but I can ping him on the community channel. The second customer talks about LiFePo4 batteries in plural, is there a specific product ID I should use or match for a family of devices?

mr-manuel, 2024-05-03:
In GUIv1 all batteries displayed the cell voltages with 3 decimals. I don't think that it should be different for some batteries. Better would be that for all batteries there are 3 decimals.

jepefe, 2024-05-03:
> Ideally would avoid product-specific variations, but ok. I don't know what battery the first user was using, but I can ping him on the community channel. The second customer talks about LiFePo4 batteries in plural, is there a specific product ID I should use or match for a family of devices?

As discussed on Slack, as a rule of thumb:
DC Voltages: 2 decimals.
AC Voltages: No decimals.
This specific case which is cell voltage: 3 decimals.

So no product specific but measurement specific.

commit ebf584f6 (2024-05-16): Improve battery details pages: show cell voltages to 3 decimals

For some battery types (particularly LiFePO4 batteries), it is
important to have 3 decimals of precision in order to detect
faulty cells, etc.

Fixes #1019