victronenergy/gui-v2 #1025: DVCC "Autoselected" BMS is always "None"

ReinvdZee, 2024-05-01:
In the DVCC menuon GUI-v2, when setting the controlling BMS to "Autoselected", the selected BMS is "None":
[image]

The path `com.victronenergy.system/ActiveBmsService` is set correctly: `com.victronenergy.battery.socketcan_can0` (or something similar). When the BMS is manually selected, the dbus path does not change, but the selected BMS is filled in correctly on the gui. 

The BMS autoselect feature works as expected on GUI-v1:
[image]

ReinvdZee, 2024-05-01:
This was reported by a battery manufacturer which recently gained support by Victron. The manufacturer reports a slightly different behavior, which I have not been able to reproduce. In fact, I do not even see the same behavior on his system while he does see it (so he reports).

As given in the issue description, I see "None" as autoselected:
[image]

The support engineer of the battery manufacturer sees:
[image]

We are both viewing the GUI remotely via VRM. He will report what the physical display shows as well.

jpetrell, 2024-05-02:
This looks like duplicate to #873.

"The problem is that the value of mqtt/system/ActiveBmsService is like a D-Bus service uid (like com.victronenergy.battery.tty0) instead of a MQTT service uid (like mqtt/battery/123). So, the composed uid becomes something like com.victronenergy.battery.tty0/ProductName, which are valid on D-Bus, but not on MQTT."

mr-manuel, 2024-05-02:
Indeed it's the same and https://github.com/victronenergy/gui-v2/pull/1032 did not fix it.

jpetrell, 2024-05-08:
Setting as blocked, @ReinvdZee promised to look into this to see if systemcalc needs changes.

ReinvdZee, 2024-05-13:
@jpetrell, [this commit](https://github.com/victronenergy/dbus-systemcalc-py/commit/8990a0430b8243ec395c6118aec5fd6d35fc00f4) should fix it, it is on the todo but not merged in venus yet.

jpetrell, 2024-05-21:
> [this commit](https://github.com/victronenergy/dbus-systemcalc-py/commit/8990a0430b8243ec395c6118aec5fd6d35fc00f4) should fix it, it is on the todo but not merged in venus yet.

Rein told me that this is Included in the beta release since v3.40~14. Removing blocked status.

jpetrell, 2024-05-23:
Assigning to @blammit, who expressed interested in tackling this and is more familiar with the topic. Shared Python scripts from Rein.

commit e3c800e5 (2024-05-02): Fix BMS name detail in DVCC settings

Fixes #1025.
commit 5ceb5e3b (2024-05-28): Fix DVCC "Auto selected" BMS name on MQTT/Wasm

The /ActiveBmsInstance path is now available from systemcalc, so use
this value to find the correct BMS service uid for fetching the BMS
service product name or custom name.

Previously this was not possible, because on MQTT both the service
name and device instance are required for locating the correct service
uid for the BMS service.

Fixes #1025
commit 798c9a29 (2024-05-28): Fix DVCC "Auto selected" BMS name on MQTT/Wasm

The /ActiveBmsInstance path is now available from systemcalc, so use
this value to find the correct BMS service uid for fetching the BMS
service product name or custom name.

Previously this was not possible, because on MQTT both the service
name and device instance are required for locating the correct service
uid for the BMS service.

Fixes #1025
closed 2024-05-28 by blammit