victronenergy/gui-v2 #1026: Overview -> Grid meter -> Page

mr-manuel, 2024-05-01:
When viewing the overview page and then clicking on the grid meter all values on the page are empty. I'm using the internal meter of the MultiPlus II as grid meter.

[image]

jpetrell, 2024-05-20:
> When viewing the overview page and then clicking on the grid meter all values on the page are empty. I'm using the internal meter of the MultiPlus II as grid meter.

Note there was a typo, the box was fixed to show "Grid", and not "Grid meter". When I open "Grid meter" in gui-v1 device list I also get settings page with "--", in state "Not connected". 

But there is something going wrong. When I tap "Grid" AC input widget I get empty page, which cannot be right either.

Looks like tapping AC input box on the OverviewPage tries to open either PageAcInModelDefault.qml or PageAcInModelGenset.qml. In Boat/Motorhome demo cases the device path passed to PageAcIn is the same as the device path when opening Inverter /Charger box, i.e. "MultiPlus-II 24/3000/70-32" device with path dbus/com.victronenergy.vebus.ttyO1. Though with AC input box case the DBus properties don't match what the settings is trying to display.

AC input settings tries to show following properties, but ends up displaying empty page because none are found:

```
/Ac/L1-3/Voltage
/Ac/L1-3/Current
/Ac/L1-3/Power
/Ac/Power
/Ac/Energy/Forward
/Ac/L1-3/Energy/Forward
```

ReinvdZee, 2024-05-21:
@jpetrell,

I think the AC input box should use `com.victronenergy.system` instead of the `vebus..` path. For example, in case of a three-phase system, there will be three MPs (each with their own path), so we can't just use the data from one of the inverters. 

Luckily, systemcalc (`com.victronenergy.system`) is there to collect the correct data. Most of the required paths are already there (like `Ac/Grid/L1-3/Power`), some of them may need to be added.

@izak, Is this correct? Or should the AC input box pull its data directly from `com.victronenergy.vebus..`?

jpetrell, 2024-05-21:
Also [reported](https://community.victronenergy.com/comments/280475/view.html) by the customer: tapping AC input box with MultiPlus-II system opens empty AC inputs page:
[image]

ReinvdZee, 2024-05-21:
@jpetrell,

The AC input box uses the data from a grid meter, this can either be the Multiplus itself or a dedicated grid meter (like Carlo Gavazzi EM24 as in the demo). 

When I load the Boat/Motorhome Demo 2, it uses the data of the CG EM24 correctly:
[image]

The paths you specify are there:
[image]

When there is no grid meter, I suppose the Multiplus should be used as grid meter. I'm not sure how this is supposed to work for three-phase systems (three Multis) without a grid meter. 

@izak Can you help us out?

ReinvdZee, 2024-05-21:
@jpetrell

Also have a look at [System Overview and mapping D-Bus paths](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit#heading=h.mlzm22he0ov) in the UI doc.

And then "step 2" in particular, that distinghuishes between a grid meter and using the multiplus as grid meter. The paths to be used to fill the AC input box are different.

blammit, 2024-05-22:
@ReinvdZee @izak If the AC input comes from a `vebus` service, as in the case above with the Multiplus, should gui-v2 just show the [VE.Bus settings page](https://github.com/victronenergy/gui/blob/master/qml/PageVebus.qml) instead when the box is clicked? That would probably make more sense, since the `vebus` service wouldn't have any of the settings for Energy, Phase Sequence etc.

blammit, 2024-05-23:
@jepefe ^ wondering what you think about this as well.

ReinvdZee, 2024-05-23:
@blammit.

It should use a grid meter if its there, and use the data from the MP if not. Since the MP is the grid meter in this case, redirecting to the VE.bus settings page may be a good choice. 

Again, not sure if and how this works with three-phases. Or maybe a three-phase system can only work with a grid meter.

jepefe, 2024-05-23:
The MultiPlus will report all the three phases.

ReinvdZee, 2024-05-23:
@jepefe, Ok nice. 

Then we should populate the AC input page with the data from vebus if the MP is selected as grid meter, as per the [UI design doc](https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit#heading=h.mlzm22he0ov): 

> For grid and for genset:
com.victronenergy.grid/Ac/NumberOfPhases
com.victronenergy.grid/Ac/L1/Voltage | VAC
com.victronenergy.grid/Ac/L1/Current | A
com.victronenergy.grid/Ac/L1/Frequency | Hz
com.victronenergy.grid/Ac/L2/Voltage | VAC
com.victronenergy.grid/Ac/L2/Current | A
com.victronenergy.grid/Ac/L2/Frequency | Hz
com.victronenergy.grid/Ac/L3/Voltage | VAC
com.victronenergy.grid/Ac/L3/Current | A
com.victronenergy.grid/Ac/L3/Frequency | Hz

>For vebus:
/Ac/ActiveIn/L1/F | Hz
/Ac/ActiveIn/L1/I | A
/Ac/ActiveIn/L1/P | W
/Ac/ActiveIn/L1/V | VAC

And indeed, `/Ac/ActiveIn/..` is there for `L2` and `L3` as well. The MP does report energy, but only for the phase it is on, so we may need to add that for the other phases as well if desired. 

Agreed? @jepefe, @izak?

closed 2024-05-24 by jpetrell

jpetrell, 2024-05-24:
AC input widget fixed to open VeBus settings if AC input widget is not a grid meter. Please check if further changes are needed, at least now it doesn't open settings that reports invalid values.

There is still #1137 open about aligning VeBus and Inverter/Charger settings pages, which show identical information in different layout.

commit d2740edb (2024-05-23): Open VeBus settings if AC input widget is a vebus device

Fixes #1026.
commit 30af6711 (2024-05-23): Open VeBus settings if tapped AC input widget is not a grid meter

Fixes #1026.
commit 21d84eff (2024-05-24): Open VeBus settings if AC input widget is not a grid meter

Fixes #1026.
PR #1136 (2024-05-24): Open VeBus settings if AC input widget is not a grid meter
Fixes #1026.