victronenergy/gui-v2 #1030: Add keyboard navigation to gui-v2

jpetrell, 2024-05-02:
In gui-v1, there is keyboard navigation when using it on Remote console. And that is used everyday by many technicians that sit behind their laptop and sort issues out or configure systems. Enable keyboard navigation on gui-v2 also.

[image]

jpetrell, 2024-05-02:
Arrow navigation e.g. between main views and vertically on the settings pages seems pretty straight-forward (i.e. the keyboard focus is moved to the adjacent item that is located in the arrow direction), and for the common UI components (dialogs, sliders, spin boxes, etc.) I can initially use gui-v1 behavior as an example.

Some open items to decide:
- How should the selected item look like and behave? Always there, or fade out after time out (as not meaningful for the touch case)? Visually mimick press effect or have it more clearly highlighted like on gui-v1 that displays blue background?
- Should there be key shortcut for opening Control cards. Could be that pressing Esc (back) on the main views would toggle between Control cards and the main carousel?
- Should we allow selecting boxes in the Overview page, or reserve left/right arrows for navigating to the adjacent Overview/Notifications pages
- How to switch between Tanks and Environment tabs in the Levels page? Clicking arrow up focuses tab bar?
- How to access status bar buttons? Similar to the tab bar, clicking arrow up focuses the first status bar button? And after clicking the right arrow key moves the focus to the status bar button on the right
- Should we show the keyboard layout somewhere like shown above in the Remote Console case?

jpetrell, 2024-06-10:
Not for 1.0.0.

jpetrell, 2024-06-19:
Updated the prototype to match Serj's design.

Todo:
- While the key navigation works on most sub-pages some of the main pages like the Control cards, Overview and Notifications pages don't support key navigation yet.
- Many settings list items that are not valid in the current customer's configuration are hidden. Take a variant of gui-v1's ObjectModel into use in gui-v2 that removes hidden settings list items from the model so they stop interfering with the key navigation
- Many settings list items (code has 234 enabled bindings) that are not allowed to be modified are currently set as disabled. Disabling also breaks the key navigation, find a way to allow navigation.
- More testing is still needed for the all the pages and settings

https://github.com/victronenergy/gui-v2/assets/2203667/a37b8646-085a-4f1b-963b-53184d4f2071

blammit, 2024-11-25:
Discussed with @chriadam that https://github.com/victronenergy/gui-v2/issues/1232 should be done first, as we need to make sure that keyboard navigation works with the redesigned settings pages.

blammit, 2024-11-25:
Depends on https://github.com/victronenergy/gui-v2/issues/1732.

MikeTrahearn-Qinetic, 2025-01-24:
Need to also consider the scenario of when a currently focused item is removed from a ListView.
I'm not quite sure of the case where the entire page is highlighted as it presupposes a navigation "depth" which may confuse the user experience and mental mapping.

closed 2025-05-12 by blammit