victronenergy/gui-v2 #1035: Gui-v2 Device list: "Remove disconnected devices" has no function

ReinvdZee, 2024-05-06:
Seen on gui-v2 v0.2.18. The device list properly shows connected devices: 

[image]

When disconnecting the device, the button "Remove disconnected devices" pops up. In gui-v1 this button then removes the disconnected devices from the list. However, in Gui-v2, the device already disappeared from the list:

[image]

So the button has no function. When pressed, the button disappears. 

I suppose either the disconnected should remain present until the button is pressed, or disconnected devices disappear immediately and the button should not appear at all.

kwindrem, 2024-05-06:
The way gui-v1 works is handy for debugging a system since the device doesn't just disappear from the device list with no reason as to why, or that it was ever there in the first place.

Having devices that go away automatically is a surprise to the end user (I thought I had solar, where did it go?)

blammit, 2024-06-18:
There are two separate issues here:

- disconnected devices are not visible in the Device List
- the Device List is rebuilt each time the page is opened, so if a device is disconnected when this page is not visible, it will not appear once the page is opened

chriadam, 2024-07-04:
#1245 was merged, can this be closed?

blammit, 2024-07-05:
Yep, closing.

chriadam, 2024-07-05:
- load cerbo, change between demo modes, so some devices will "vanish"
- those devices should be considered "disconnected" - they should still show up (as disabled/grey) in device list
- then there should be a "remove" button to remove them.

DanielMcInnes, 2024-07-05:
If we change demo mode from 'boat motorhome demo 1' to 'ESS', gui-v1 shows the old devices (eg. refrigerator) that are no longer connected. On gui-v2, we don't see the disconnected devices in the device list.

DanielMcInnes, 2024-07-05:
[image]

blammit, 2024-07-11:
Discussed in sync meeting that when viewing the Device List from wasm, the user will only see disconnections that have occurred during runtime. In the long term there will need to be a solution provided by the lower levels to allow gui-v2 to also see disconnections that occurred before the loading of the wasm app, so that the displayed list looks exactly like the one on the device, even if the wasm app has not been running for the same amount of time.

closed 2024-07-19 by blammit