victronenergy/gui-v2 #1041: Improve UI for AC in/out when feeding back to grid

blammit, 2024-05-09:
When ESS feed-in is enabled, negative power for the AC in/out indicates that the AC is feeding back to the grid. In this case, gui-v2 can:

- drop the minus sign in the Brief and Overview pages
- show the direction icon in the Overview box; this is already present in the Brief gauge
- use zero as the minimum when feed-in is disabled

It's possible for negative values to occur even when feed-in is disabled, but in this case it's just noise, and the minus sign can be removed.

[image]
[image]

blammit, 2024-05-09:
For reference, drop the minus signs here:

[image]
[image]

mr-manuel, 2024-05-10:
Is this already somewhere in a branch with the arrow and green color?

mr-manuel, 2024-05-10:
Is something similar also planned for the Inverter/Charger box for noting the difference between inveting and charging?

blammit, 2024-05-13:
> Is this already somewhere in a branch with the arrow and green color?

Yes, this is already done for the Brief page in `main`.

blammit, 2024-05-13:
> Is something similar also planned for the Inverter/Charger box for noting the difference between inveting and charging?

I don't think there is at the moment. Perhaps @jepefe can comment more on whether that will happen.

mr-manuel, 2024-05-13:
Strange, I'm using the `v0.2.19`, which is the latest main build. Has it to be enabled?

[image]

mr-manuel, 2024-05-13:
Now I discovered, that it's only shown, if `AC-coupled PV - feed in excess` AND `DC-coupled PV - feed in excess` is enabled. Shouldn't it be enabled if  `AC-coupled PV - feed in excess` OR `DC-coupled PV - feed in excess` is enabled?

blammit, 2024-05-13:
I'm not sure about that criteria, but it's currently shown when power < 0 and:

com.victronenergy.system/Hub = 4
com.victronenergy.settings/Settings/CGwacs/OvervoltageFeedIn = 1
com.victronenergy.settings/Settings/CGwacs/PreventFeedback = 0

mr-manuel, 2024-05-13:
`com.victronenergy.system/Hub = 4` --> Checks if the system has ESS
`com.victronenergy.settings/Settings/CGwacs/OvervoltageFeedIn = 1` --> `DC-coupled PV - feed in excess = Enabled`
`com.victronenergy.settings/Settings/CGwacs/PreventFeedback = 0` --> `AC-coupled PV - feed in excess = Enabled`

I think here is to clarify, if `OvervoltageFeedIn ` AND `PreventFeedback ` needs to be met or only one of them. In my eyes only one of this two is needed.

jepefe, 2024-05-13:
@blammit 

```
com.victronenergy.system/Hub == 4 && (com.victronenergy.settings/Settings/CGwacs/OvervoltageFeedIn == 1 || 
com.victronenergy.settings/Settings/CGwacs/PreventFeedback == 0)
```

jepefe, 2024-05-13:
> > Is something similar also planned for the Inverter/Charger box for noting the difference between inveting and charging?
> 
> I don't think there is at the moment. Perhaps @jepefe can comment more on whether that will happen.

The battery has a label that says 'Charging' or 'Discharging', so it must be clear enough on the current implementation. Or is it not?

mr-manuel, 2024-05-13:
It is clear. It would been only for consistency, if one gauge turns green when negative, it would be obvious that also the other has the same behaviour.

On the overview page you have charging and discharging, but not on the brief page.

blammit, 2024-05-15:
Thanks, will make that fix as part of these changes.

closed 2024-05-15 by blammit

blammit, 2024-05-15:
> It is clear. It would been only for consistency, if one gauge turns green when negative, it would be obvious that also the other has the same behaviour.

> On the overview page you have charging and discharging, but not on the brief page.

@jepefe what do you think? Should the AC load bars turn green when the system is charging? If so, please let me know which `/SystemState/State` values should be matched for the "charging" case.