victronenergy/gui-v2 #1047: Clearly indicate text field confirmation step

jpetrell, 2024-05-13:
Tap a text field to focus the editor. Edit the text using a keyboard. Continue to the next task at hand.

Actual result: If you don't remember to press enter key on the virtual or physicalkeyboard the changes are not committed. This is inconsistent to other inline components like switches and sliders. And in comparison in dialog cases the commit and cancel steps are clear.

I was surprised when I saw this behavior initially, and I still lose my changes from time to time.

Expected result: Unless the value changes implicitly (e.g. when toggling a switch or moving a slider handle), clear cancel and confirmation buttons are displayed (like how e.g. [gui-v1](https://private-user-images.githubusercontent.com/2203667/328180851-3e5f594f-9a2f-495e-b99e-ff3bb5be50ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU1ODgwNTQsIm5iZiI6MTcxNTU4Nzc1NCwicGF0aCI6Ii8yMjAzNjY3LzMyODE4MDg1MS0zZTVmNTk0Zi05YTJmLTQ5NWUtYjk5ZS1mZjNiYjViZTUwZWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTNUMDgwOTE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTQwODcwNjJhOTE0YmQxNWNkZWI4ZGFiMzUyYzEyYTRjYzlhNTk3MTJlN2JmYzU3NmNiOWE5ZWY5YmM3NjE0NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.8GyuCkhVoUZ_hOZzmFdZcTOLixyL4tcFyGul93LYI6M) does) so the user knows they need to manually confirm the action for it to take effect.

This is a suggestion, needs discussion and design before proceeding to any changes.

jpetrell, 2024-05-17:
This is maybe less of an issue with GX devices where you usually press enter to close the keyboard, ending the editing session. On browser (wasm) version there is no visible button on the GUI to commit.

blammit, 2024-05-20:
Discussed that the text field could attempt to save changes as soon as it loses focus.

jpetrell, 2024-05-30:
FYI @mr-manuel filed similar issue here https://github.com/victronenergy/gui-v2/issues/1167.

mr-manuel, 2024-07-15:
Any update on this? Should be at least `medium_prio` and pre `v1.0.0` release. At least regarding the feedback when saving the data, see https://github.com/victronenergy/gui-v2/issues/1167.

mpvader, 2024-07-15:
I’ve added it to the milestone and set the priority as requested, assuming that at least giving some feedback is trivial.

 @chriadam , @blammit , @DanielMcInnes please do as you think is best!

blammit, 2024-07-16:
Discussed previously with Joona that:

- the use of validators or anything that modifies the input directly provides a less ideal user experience; it would be better to allow the user to input text/numbers freely, and then validate it later (e.g. when the input field loses focus)
- if the text is invalid, should show some info/error to indicate what is wrong with the input, instead of just reverting it and losing the current input

So ideally, on focus loss on a text field:

- if text is acceptable, then save to backend
- if not acceptable, show the error message (and colour the text red) and do not save

I will investigate the changes required to implement this consistently across the current text fields.

mr-manuel, 2024-07-16:
Why is there no feedback planned, that the data was saved? It's really strange, if you enter a value and nothing happens.

Please not only on focus loss, but also when the user hit the enter key.

blammit, 2024-07-16:
It will also happen on focus loss; the problem at the moment is that it only happens when the Enter key is pressed.

mr-manuel, 2024-07-16:
Thanks for clarification!

blammit, 2024-07-25:
Here are the current uses of ListTextField:

- ListIpAddressField: for entering an IP address (PageSettingsTcpIp.qml, PageSettingsModbusAddDevice.qml, PageSettingsFroniusSetIpAddresses.qml, PageSettingsFroniusShowIpAddresses.qml)
    - inputMethodHints: Qt.ImhDigitsOnly
    - validator: RegularExpressionValidator { regularExpression: /[0-9\.]{1,15}/ }
- ListPortField: for entering a port number (PageSettingsFronius.qml)
     - inputMethodHints: Qt.ImhDigitsOnly
     - validator: RegularExpressionValidator { regularExpression: /[0-9]{1,5}/ }
- ListQuantityField: for entering a quantity (PageSettingsDisplayMinMax.qml)
    textField.validator: DoubleValidator { decimals: root.decimals; locale: Units.numberFormattingLocaleName }
    textField.inputMethodHints: Qt.ImhFormattedNumbersOnly
- ListRadioButtonGroup: for entering a password
    - password is validated when accepted
    - text is cleared when focus is lost
- PageDeviceInfo: for entering custom name
    - maximumLength: 32
- PageGeneratorRuntimeService: for entering service interval and accumulated total offset
    - inputMethodHints: Qt.ImhDigitsOnly
    - maximumLength: 6
    - converts entered hours value into seconds when saving
- PageSettingsBatteries: for entering battery configuration name
    - maximumLength: 32
- PageSettingsBluetooth: for entering pin code
    - inputMethodHints: Qt.ImhDigitsOnly
    - maximumLength: 6
- PageSettingsGsm: 
    - for entering pin (maximumLength: 35)
    - for entering APN (maximumLength: 50)
    - for entering user name and password
- PageSettingsModbusAddDevice
    - for entering port     
        validator: RegularExpressionValidator { regularExpression: /[0-9]{1,5}/ }
        inputMethodHints: Qt.ImhDigitsOnly
        checks port is between 0-65535
    - for entering unit number
        validator: RegularExpressionValidator { regularExpression: /[0-9]{1,5}/ }
        inputMethodHints: Qt.ImhDigitsOnly
    Note these are only validated when the "add" action is confirmed.
- PageSettingsSystem: for entering name
- PageSettingsTcpIp: for entering password
    - maximumLength: 63
    - on accept, saves the value as a json-encoded value
- PageVrmDeviceInstances: for changing device instance
    - inputMethodHints: Qt.ImhDigitsOnly
    - when accepted, the input may be reverted if another device with this instance exists
- PageRsSystem: for entering custom name

The uses in PageSettingsGeneral (for entering root password) and PageSettingsRemoteConsole (for entering password) can be ignored as these will be removed.

blammit, 2024-07-25:
https://github.com/victronenergy/gui-v2/pull/1356 updates the text field implementations as proposed.

commit 606a9467 (2024-07-25): Validate and accept/reject text input when focus is lost

Align validation/accept/reject behavior of various text fields, so
when text field focus is lost:

- Do text validation if required. If it succeeds, save the input; it
it fails, highlight the text field and show a toast notification to
indicate why the input is invalid.
- If no validation is required, save the input immediately.

If the text field is accepted, then validate and save the input, and
remove the focus.

Avoid setting the "validator" property for text fields, so that text
can be freely entered before the input is accepted.

 [... cut]
commit 787d3cca (2024-07-26): Validate and accept/reject text input when focus is lost

Align validation/accept/reject behavior of various text fields, so
when text field focus is lost:

- Do text validation if required. If it succeeds, save the input; it
it fails, highlight the text field and show a toast notification to
indicate why the input is invalid.
- If no validation is required, save the input immediately.

If the text field is accepted, then validate and save the input, and
remove the focus.

Avoid setting the "validator" property for text fields, so that text
can be freely entered before the input is accepted.

 [... cut]
commit 3006b884 (2024-07-31): Validate and accept/reject text input when focus is lost

Align validation/accept/reject behavior of various text fields, so
when text field focus is lost:

- Do text validation if required. If it succeeds, save the input; it
it fails, highlight the text field and show a toast notification to
indicate why the input is invalid.
- If no validation is required, save the input immediately.

If the text field is accepted, then validate and save the input, and
remove the focus.

Avoid setting the "validator" property for text fields, so that text
can be freely entered before the input is accepted.

 [... cut]
commit 332eaec2 (2024-07-31): Validate and accept/reject text input when focus is lost

Align validation/accept/reject behavior of various text fields, so
when text field focus is lost:

- Do text validation if required. If it succeeds, save the input; it
it fails, highlight the text field and show a toast notification to
indicate why the input is invalid.
- If no validation is required, save the input immediately.

If the text field is accepted, then validate and save the input, and
remove the focus.

Avoid setting the "validator" property for text fields, so that text
can be freely entered before the input is accepted.

 [... cut]
closed 2024-07-31 by blammit