victronenergy/gui-v2 #1061: With 5 side panel widgets the last widget gets clipped by the nav bar

jpetrell, 2024-05-16:
[image]

blammit, 2024-05-16:
Not sure what we want to do here. Fade out the last widget and then fade it back in when the nav bar disappears?

kwindrem, 2024-05-16:
If you can keep the nav bar visible then the side panel should either compress as happens with the Overview page or allow scrolling vertically to expose additional widgets.

If the nav bar always disappears after a period of inactivity then what happens in this scenario is of less importance. I'd just leave it the way it is.

jpetrell, 2024-05-20:
Serj: "Dimensions and spacing are not respected in implementation. 5 widgets should fit"
[image]

commit be654f78 (2024-05-22): BriefSidePanel: reduce quantity label size

This ensures all widgets fit within the side panel height when the
nav bar is open, even when there are five widgets visible in the side
panel.

Fixes #1061
closed 2024-05-22 by blammit