victronenergy/gui-v2 #1068: Brief page: Solar yield graph PV inverter

mr-manuel, 2024-05-17:
Would it be possible to add a graph (like the AC Loads have) to the solar yield, if there is only a PV inverter (AC) without solar charger (DC)? If there is both leave it as it is and only display the solar charger graph.

[image]

blammit, 2024-05-22:
Currently systemcalc doesn't provide min/max values for PV inverters in order to draw this graph, but I assume it should be possible to provide those values if this feature was desirable.

Will leave it to @jepefe to comment on whether/when this feature might be possible.

mr-manuel, 2024-05-22:
The PV itself provides this value. See https://github.com/victronenergy/venus/wiki/dbus#pv-inverters

```
/Ac/MaxPower           <- Max rated power (in Watts) of the inverter
```

jepefe, 2024-05-22:
@blammit 
`/Settings/Gui/Gauges/Pv/Power/Max` works for both PV Inverters and Chargers.  In case that there are only PV inverters in the installation, the max value is also tracked.

blammit, 2024-05-22:
Okay, thanks! Will look into it.

blammit, 2024-05-27:
@jepefe what if there are multiple PV inverters? Should the graph show an average power value, or should we omit the graph in this case?

Also, we can show the one or 3-phase horizontal bar gauge below the graph, if there are only PV inverters. Same question applies here: if there are multiple PV inverters, would the horizontal bars show the average power value across all PV inverters, or only show it when there is a single PV inverter?

mr-manuel, 2024-05-27:
Why not simply sum the values like for the solar yield on the bottom keft of the brief page?

mpvader, 2024-06-13:
the Solar graph on the side panel, which doesn't distinguish between type (PV Inverter or MPPT Solar Charger) nor the location, should for its data calculate the total of all available solar.

gui-v2 can take the data that it needs to sum from systemcalc, see here: https://docs.google.com/document/d/1yzHSpZoeZ0WOJJPhjivbhhD_iz37PaZtnZ9GpqVPh2M/edit#heading=h.pimi6y1rfdhq

blammit, 2024-06-27:
Needs more discussion about the graphs to be displayed here.