victronenergy/gui-v2 #1069: Investigate single deployment with editable qml

DanielMcInnes, 2024-05-17:


DanielMcInnes, 2024-05-22:
Currently, gui-v2 is installed to /opt/victronenergy/gui-v2/, along with all qml files. By default, the cerbo root partition is mounted 'read only', so a remount is needed in order to edit the qml source: `mount -o remount,rw /` .

The qml source files are editable on the cerbo, and after a restart, enthusiasts will be able to see the results of their changes.

chriadam, 2024-05-22:
By default, it will load files from the filesystem, right now.  In future, it would be nice if it could do some sort of checksum to determine if a given file has changed, and if not, load the type from the qrc (as hopefully that means the engine can use the cached typedata).

DanielMcInnes, 2024-06-17:
closing, this is done

closed 2024-06-17 by DanielMcInnes