victronenergy/gui-v2 #1071: Allow custom QML to be inserted into gui-v2

blammit, 2024-05-17:
Allow users to load custom QML content into gui-v2.

For example, for:

- settings pages - add new settings
- control cards - add new cards
- main app swipe view - add new pages and entries in navigation bar
- device list - add more entries
    - would need to rework device list setup a bit to make this work

gui-v2 could load details from some json configuration. E.g. to customize the settings page, load some json file that defines the title and url to be loaded.

Initial work is to do some investigation and prototyping.

blammit, 2024-05-27:
# Deploying and loading 3rd-party QML files on device 

We want an option where 3rd-party QML files can be easily deployed to device and loaded by gui-v2. Some minimal architecture can be added to load these QML files to restrict the loading to specific pages; for example, to load some custom settings from the Settings page, or add a new page that can be launched from the navigation bar.

Requirements:

- QML deployment should be simple (e.g. scp to device, without rebuilding gui-v2)
- QML files must be accessible via device on-screen, and via the browser (i.e. wasm)

## Possibilities for loading 3rd-party QML

1. Import a QML file from a local filesystem URI (e.g. via a Loader or via `import "/some/dir`)

- Won't work as files on the device are not accessible via local file paths. Since Qt 6, QML files are compiled into the binary using `qrc` files, rather than dynamically loaded via the filesystem, unless the application has been built with specific cmake flags that specify otherwise.
- System file paths are not accessible to a wasm deployment.

2. Import a QML file from a http path (e.g. via a Loader or via `import "http://somewhere/Type.qml`)

- If files are on a different server, this would fail cross-domain resource checks. Also it's too much work for users to have their own hosting solutions.
- Device server could serve files from the same directory where the wasm bundle is located, but it seems less than ideal to be serving arbitrary QML files. 

3. Import a QML file from generated localsettings

Provide two-step deployment where QML files are deployed to device, then the user can run some binary on device to read the files and load them into local settings as raw QML code, that can then be loaded via D-Bus or Wasm.

This option is not at all feasible, and only listed here for completeness. It's not scaleable to store the data into localsettings when the QML is non-trivial, and Qt.createQmlObject() is highly inefficient.

4. Import a QML file from generated RCC

Provide two-step deployment where QML files are deployed to device, then the user can run some binary on device to read the files and generate a Qt `rcc` file that includes all 3rd-party QML types. Then, gui-v2 can register the `rcc` at runtime using QResource::registerResource(), and load the types as needed.

- Would need to package and distribute Qt `rcc` tool with VenusOS so that it is available to gui-v2. Maybe make it available to MQTT via venus-platform.
- Could have some management settings. E.g. in Settings > Third-party applications, have "Reload QML files" and that could run rcc on the device to generate a new rcc file.

blammit, 2024-05-27:
We need a solution that allows 3rd-party QML files to be accessible both 1) on device and 2) from wasm. 1) is easy because we can load files from the local filesystem, but 2) is more difficult as the wasm environment is sandboxed and you cannot access local files.

One possibility is for gui-v2 on device (or venus-platform) to read QML files and JSON configurations from a special directory on device and publish the details to some specific D-Bus/MQTT paths, to make them accessible to gui-v2 on both device and wasm.

When publishing the custom data, QML can be inserted as raw strings, and images as base64-encoded data. Alternatively the data could all be inserted as binary data and then extracted/decoded by gui-v2.

mr-manuel, 2024-05-27:
Would be great if with a JSON settings file templates could be used (refer to https://community.victronenergy.com/questions/245056/venus-os-modifying-gui-v2.html?childToView=285975#comment-285975). Like:

```JSON
    {
        "add": {
            "settings/display-and-language": [
                {
                    "title": "CustomSwitch",
                    "type": "switch",
                    "dbus-path": "com.victronenergy.settings/Custom/Path/Switch"
                },
                {
                    "title": "CustomInput",
                    "type": "textbox",
                    "dbus-path": "com.victronenergy.settings/Custom/Path/Text"
                },
                {
                    "title": "CustomSelect",
                    "type": "select",
                    "dbus-path": "com.victronenergy.settings/Custom/Path/Select",
                    "options": [
                        {
                            "value": "1",
                            "text": "Option 1"
                        },
                        {
                            "value": "2",
                            "text": "Option 2"
                        }
                    ]
                },
                {
                    "title": "CustomPage",
                    "type": "page",
                    "content": [
                        {
                            "title": "CustomSwitch",
                            "type": "switch",
                            "dbus-path": "com.victronenergy.settings/Custom/Path/Switch"
                        }
                    ]
                },
                {
                    "title": "CustomPage",
                    "type": "page",
                    "qml": "path-to-qml-file.qml"  // requires more maintenance of the plugin developer
                }
            ]
        },
        "remove": [
            "settings/relay/relay-1-on",
            "settings/some-other-item-or-page"
        ]
    }
```

jpetrell, 2024-05-27:
I would limit JSON to a simple configuration file that points out where can the QML, translations and other assets be found.

> Would be great if with a JSON settings file templates could be used

There has been such systems before, but overall feels silly to introduce a new settings UI declaration syntax in JSON when you can can assemble such a simple settings page with QML already, without the need to implement and support custom JSON schemas and formatters. I would consider declaring settings pages with JSON if we need to somehow limit what 3rd party "apps" are allowed to do. 

```
JSON 
+ JSON is ubiquitous
+ Security, can more easily limit what such settings page is allowed to do
- More limited what kind of use cases can be accomplished
- Need to implement a new JSON schema/formatter/validator that supports all the needed elements and attributes
- No translation support
- Static declaration
```

```
QML
+ Actively used and tested internally
+ Optimized for writing GUI declarations
+ Access to the latest gui-v2 UI components
+ Supports translations out of the box
+ Not limited to simple forms, can do more custom use cases and layouts
- Not so much used outside embedded circles
- Supports advanced features that may confuse inexperienced developers
```

How you would write the above settings JSON currently in QML:

```
Page {
  GradientListView {
    model: ObjectListModel {
      ListSwitch {
        text: "CustomSwitch"
        dataItem.uid: "com.victronenergy.settings/Custom/Path/Switch"
      }
      ListTextField {
        text: "CustomInput"
        dataItem.uid: "com.victronenergy.settings/Custom/Path/Text"
      }
      ListRadioButtonGroup {
        text: "CustomInput"
        dataItem.uid: "com.victronenergy.settings/Custom/Path/Text"
        optionModel: [
          { display: "Option A", value: 1 },
          { display: "Option B", value: 2 }
        }
      }
      ListNavigationItem {
        text: "CustomPage"
        onClicked: Global.pageManager.pushPage("com.victronenergy.settings/Custom/Path/Switch")
      }
    }
  }
}
```

Which could be further simplified with few syntax improvements:

```
SettingsPage {
  ListSwitch {
    text: "CustomSwitch"
    uid: "com.victronenergy.settings/Custom/Path/Switch"
  }
  ...
}
```

mr-manuel, 2024-05-27:
If you currently take a look at the scripts/packages that are available from the [SetupHelper](https://github.com/kwindrem/SetupHelper) you will see that it is a lot of effort maintaining a project for different Venus OS versions, since for every different version a different QML is needed. This increases the maintenance for the package developer enormous. If we could use templates that are consistent indipendent of the version that would be very helping. If Victron changes something in the QML structure, it's automatically changed also in the template. This would mean the developers are not in a rush to update their packages (like [Kevin](https://github.com/kwindrem) does) because a new version was released any may cause a white screen for a lot of users.

Therefore it does not need to be a JSON file, but please provide the possibility to provide only settings without a QML structure for simple projects, where the QML is then generated from the settings file.

mr-manuel, 2024-05-27:
> - No translation support

This is not true, since you could also provide the different languages in the JSON and fallback to english, if a language is not provided.

kwindrem, 2024-05-27:
I have a program (UpdatePackage) that scans a database Venus OS versions and indicates where file changes are needed. So not every file for every VenusOS version is needed. But yes it can be a lot of work to modify files for a new Venus OS version. I have a few ways around that, the most recent is the ability to patch files rather than replacing them with a version modified by the author. This reduces but does not eliminate the need for multiple versions of the same file.

mpvader, 2025-08-21:
Notes as discussed:

- the json won't go into localsettings aka com.victronenergy.settings. Instead, venus-platform, or another service - tbd, will put the json data out on dbus (and MQTT) in some place; doesn't really matter where and paths will be defined once someone starts on this part of the work. That service will read it from disk and put it on dbus.
- `UI Customisations` header will be renamed to `Installed third party apps `. Why? Because the switches don't only control the customisations, they enable/disable the whole app.
- Wrt the device list customisations it will be made possible to provide a list of productids.

Then, we discussed how to lower the bar for users to get started on this. Ie. take an example provided by us to make modifications, change that and see their changes reflect in gui-v2. For example:

1. copy folder structure (example1 into my-mod)
2. start changing qml files in there
3. not have to instal an SDK or run a compiler or something else on his host pc, ssh-ing into a device and doing things there should be sufficient.

Chris checked, and the tooling needed to build the "all-in-one" file, containing images, qml files, configuration, translations, is no more than max a few megabytes. So can be installed on the GX device. 

More details about that tooling in `tools/customisationcompiler.py` which is part of the PR linked above by Chris

Next step is for Matthijs to:
1. discuss this further with Manuel and see how to get further feedback
2. get the over-arching plan progressed

closed 2025-11-06 by chriadam