victronenergy/gui-v2 #1073: Brief page: Side panel grid graph fill

mr-manuel, 2024-05-17:
The grid graph should be filled to zero and not to the bottom.

Reference to https://github.com/victronenergy/gui-v2/pull/1065#issuecomment-2117300315

## Current
[image]

## Expectation
[image]

blammit, 2024-05-22:
@jepefe any thoughts on this? The current gui-v2 design uses a graph design that is different from the one in VRM portal.

kwindrem, 2024-05-23:
Yes all graphs and gauges should fill from zero like is shown in the expectation. Filling to the minimum displayed on the graph/gauge doesn't make any sense.

jepefe, 2024-05-23:
@blammit Is it is not too much work, please change it the filling to match vrm.

blammit, 2024-05-27:
@DanielMcInnes do you want to look into this one, since you're more familiar with the LoadGraph?

PR #1226 (2024-06-17): AcInputs load graph on side panel should be zero-centered
Grid draw should be shown in blue above the midpoint (zero amps) dotted line, and grid feed-in should be shown in green below the midpoint dotted line.

Fixes #1073
commit 886896a4 (2024-06-17): AcInputs load graph on side panel should be zero-centered

Grid draw should be shown in blue above the midpoint (zero amps)
dotted line, and grid feed-in should be shown in green below the
midpoint dotted line.

Fixes #1073
closed 2024-06-17 by chriadam