victronenergy/gui-v2 #1075: improve deciding to use grid parallel overview

DanielMcInnes, 2024-05-19:
gui-v1 commit: 8794e5aed69dbeeb3b3dae246bbac4b39d7a5fdb

jpetrell, 2024-05-29:
> In the past, OverviewGridParallel was used for any system with type Hub-4 or ESS
> With the addition of other hardware with the same capabilities, a better way is needed

Not sure what should be done here in gui-v2 scope. There is no equivalent of OverviewGridParallel in gui-v2, though there are bunch of places where com.victronenergy.system/SystemType is checked for values "ESS" and "Hub-4" in ESS card visibility and ESS/Hub-4 settings, but gui v1 has the same checks so seems those are good.

These are the related issues: 
- [#1025](https://github.com/victronenergy/venus/issues/1025) Show state of whole system when grid meter installed even if not used by ESS   
- [#301](https://github.com/victronenergy/venus-private/issues/301) Multi RS three phase with EM: fix visualisation of non-critical loads

@ReinvdZee any idea what is needed to adapt gui-v2 Overview page when "when grid meter installed even if not used by ESS"? Or "fix visualisation of non-critical loads" for Multi RS three phase?

kwindrem, 2024-06-03:
The best overview is difficult to choose automatically. In GuiMods, I allow the user/installer to select the appropriate overview. A better approach is what is currently being done in gui-v2 where the tiles displayed is based on which components exist in the system.

jpetrell, 2024-06-04:
Discussed in dev meeting that we can close this one for gui-v2, there is no direct equivalent use case.

closed 2024-06-04 by jpetrell