victronenergy/gui-v2 #1077: add support for RS system

DanielMcInnes, 2024-05-19:
gui-v1 commit: 8981fa9bc8ac55f559b7dba569d9af5d7332c26c

mpvader, 2024-06-04:
@izak , on gui-v2, on the overview page, there is an action when the user clicks the inverter/charger icon in center top.

Is this a good rule for @blammit to implement?

1. if multiple AC systems and/or Multis, show a list of VE.Bus + AC Systems
2. if only one VE.Bus -> no list, but VE.Bus page opens
3. if only one AC system -> no list, but AC system page opens
4. if only one RS unit -> no list, and also no AC system, but RS page opens.

izak, 2024-06-04:
There is always an acsystem, even if there is only one Multi-RS, so rule 4 is not necessary.

Also, systems with plain inverters (not inverter/chargers) probably need to be accommodated? So can probably be consolidated to:

1. If Multiple AC-systems, Multis or other inverters, show a list
2. If only AC-system, VE.Bus or Inverter, open the relevant page directly.

blammit, 2024-06-05:
As part of this integration in gui-v2, discussed that when the Inverter/Charger overview widget is clicked, instead of 
showing a list of these services:

- inverter
- vebus 
- multi

Show these instead:

- inverter
- vebus
- acsystem

And any multi services can be accessed via the acystem instead.

mpvader, 2024-06-05:
@izak for my item 4: what is the advantage of showing acsystem over hiding it for the user if he has just one RS?

Having AC system adds levels & complexity in the UI

izak, 2024-06-05:
@mpvader If it is just about simplifying the UI, then indeed, there is no reason to show the acsystem instead of the RS when there is only one. I don't have a preference here.

mpvader, 2024-06-05:
@izak Should Bea “hide” the ac system in that case? Or you take it out of the dbus?

mpvader, 2024-06-05:
understand @izak ! thanks.

@blammit all clear to you now?

blammit, 2024-06-06:
Okay, so I think it is as per my previous comment -- when clicking the Inverter/Charger box, we will show a list of these:

- inverter
- vebus
- acsystem

And the user will click on the acsystem to get to the RS devices.

The control cards still show the RS devices rather than acsystems, i.e. there is one control card for each RS device (along with other control cards for `inverter` and `vebus` devices).

mpvader, 2024-06-06:
Hi Bea, yes but show the list only of there is multiple entries after compiling the list. If there js just one entry (which is the case for 99% of all systems) then I want to the user to go straight to that entry.

and, I think you know this but just to make sure: clicking the AC system entry takes the user to the AC System page. Which has somewhere in its submenus a list of Multi RS-es.

Control cards: one per AC system is needed. Not one per Multi.

blammit, 2024-06-07:
> Hi Bea, yes but show the list only of there is multiple entries after compiling the list. If there js just one entry (which is the case for 99% of all systems) then I want to the user to go straight [...]

Yes, this is the implemented solution 👍 If there is only one item, then we don't show the list. (Sorry, should have clarified in the previous comment.) And yes, clicking the AC system entry goes to the AC system page, which contains the "Devices" entry with a list of Multi RS devices.

> Control cards: one per AC system is needed. Not one per Multi.

That makes sense. Fixed this now in https://github.com/victronenergy/gui-v2/pull/1199

chriadam, 2024-06-20:
As per testing on Wednesday:
1) Could not change inverter mode via control card, but could via device list.  The mode was shown correctly in the control card if it was changed via device list, but could not be changed via control card.  Tested with "GR 3phase without EM".

commit 3320835c (2024-06-05): Add `acsystem` and `multi` settings and Device List entries

Add PageRsSystem, PageMultiRs and associated pages.

The Device List now shows 'acsystem' services instead of 'multi'
services. Clicking on an acsystem list item opens PageRsSystem, which
provides a 'Devices' list item to navigate to a list of multi services
for that RS system; each multi service navigates to the PageMultiRs
for that service.

Fixes #1077, #1142, #1093
commit cd6d107c (2024-06-05): Add `acsystem` and `multi` settings and Device List entries

Add PageRsSystem, PageMultiRs and associated pages.

The Device List now shows 'acsystem' services instead of 'multi'
services. Clicking on an acsystem list item opens PageRsSystem, which
provides a 'Devices' list item to navigate to a list of multi services
for that RS system; each multi service navigates to the PageMultiRs
for that service.

Fixes #1077, #1142, #1093
closed 2024-07-01 by blammit