victronenergy/gui-v2 #1080: add onEditAttemptWhenDisabled signal

DanielMcInnes, 2024-05-19:
gui-v1 commit: ea362af163cc4528315b52656a175bd71581a1b1

blammit, 2024-05-28:
I think this one can be handled a bit differently in gui-v2, instead of porting it directly.

In gui-v2 there are some UI indicators for when items are disabled, and there is space below an item to explain why it's disabled, so it would be nice  toshow a caption below an item explaining why it cannot be changed, instead of waiting for the user to click it (and see it can't be clicked) to show this information.

blammit, 2024-05-28:
In https://github.com/victronenergy/gui-v2/pull/1155, have added explanation text below the text items instead of showing it in onEditAttemptWhenDisabled().

blammit, 2024-05-29:
In MbSwitchForced, it shows "Forced off" or "Forced on" when the switch cannot be changed, so this seems sufficient without needing an extra caption to indicate that the setting cannot be changed.

closed 2024-05-29 by blammit