victronenergy/gui-v2 #1086: Update notifications to match gui-v1

DanielMcInnes, 2024-05-19:
TODO - investigate if this is actually necessary. Gui-v1 works a bit differently.

gui-v1 commit: 88c5a7073249bec0b4bfe2ba0a955b90a107bf04
    Let the status tile look for notifications in dbus
    
    Notifications were migrated from gui to venus-platform.
    Let gui use the paths created by venus-platform to display
    the active notifications properly in the status tile.

jpetrell, 2024-05-31:
AFAICS the [commit](https://github.com/victronenergy/gui/commit/88c5a7073249bec0b4bfe2ba0a955b90a107bf04) migrates notifications implementation to to use venus-platform, which is already the case with gui-v2. @ReinvdZee was this straight-forward migration or were there also some behavioral changes included?

ReinvdZee, 2024-05-31:
@jpetrell 

This was a straightforward migration to let `gui` use the `/Notifications/..` paths provided by `venus-platform`. If it works properly now, then no changes are required to gui-v2.

jpetrell, 2024-06-04:
>  If it works properly now, then no changes are required to gui-v2

Cool. Closing. :+1:

closed 2024-06-04 by jpetrell