victronenergy/gui-v2 #1087: PageBatteryDetails: hide unused items

DanielMcInnes, 2024-05-19:
gui-v1 commit: 8118d8f243bef995b5c8127d5520417668144c82

chriadam, 2024-06-20:
As per testing on Wednesday:
1) Separator should be hidden from e.g. "Lowest cell voltage" etc list items, if the cell id is unknown.
2) If no battery details are visible / valid, we should hide (or at least disable) the "Details" drilldown in battery device page.

blammit, 2024-07-12:
> If no battery details are visible / valid, we should hide (or at least disable) the "Details" drilldown in battery device page.
This is fixed by https://github.com/victronenergy/gui-v2/issues/1201

blammit, 2024-07-16:
> Separator should be hidden from e.g. "Lowest cell voltage" etc list items, if the cell id is unknown.

This is of a lower priority and takes more work to resolve due to the way that ListQuantityGroup is built at the moment. I have a created an issue with some related remarks [here](https://github.com/victronenergy/gui-v2/issues/1338), so we can close this task as the main issue has been resolved.

commit 8966e1f5 (2024-05-31): Hide unused items on the battery details page

Fixes #1087.
PR #1164 (2024-05-31): Hide unused items on the battery details page
Fixes #1087.
commit 6715c6b9 (2024-05-31): Hide unused items on the battery details page

Fixes #1087.
closed 2024-07-16 by blammit