victronenergy/gui-v2 #1093: "MultiRS" and "charger" settings pages are not yet implemented

DanielMcInnes, 2024-05-19:
I'm not sure why these are still hooked up to PageNotYetImplemented.qml in DeviceListPage

blammit, 2024-05-27:
There was a discussion in https://github.com/victronenergy/gui-v2/issues/328 that these were not required for implementation yet. But maybe that has changed? Need to confirm with @mpvader

mpvader, 2024-05-27:
Hey, yes its changed - or at least about to. Izak is finishing up his implementation of it in gui-v2.

blammit, 2024-05-28:
@mpvader did you mean gui-v2? I checked with @izak and he said he has implemented updates in gui-v1. I can look at porting those to gui-v2.

mpvader, 2024-05-28:
Yes, good catch! Amended. Pls do

blammit, 2024-06-10:
Changes merged.

chriadam, 2024-07-02:
https://github.com/victronenergy/gui-v2/pull/1292/files fixes this issue according my testing

chriadam, 2024-07-04:
Fixed.

closed 2024-07-04 by chriadam