victronenergy/gui-v2 #1097: gui-switcher: only allow starting gui-v2 if there is a LCD

DanielMcInnes, 2024-05-19:
gui-v1 commit: 2c61aa6a408f65949ba8b5f3a0d6d4be8cf5d0a1

jpetrell, 2024-05-29:
> The gui-v2 native version is only shown on an actual LCD, the webassembly version is used for browser access. So don't show an option to switch to gui-v2 if there is no LCD, since it won't work. The only thing it does is making gui-v1 unavailable and there is no way to undo that.

There is no equivalent case here for gui-v2, but if wanted we could hide the "Onscreen UI (GX Touch & Ekrano)" setting on gui-v2 wasm build, but that requires that Venus platform exposes /DisplayPresent since the wasm build cannot use the `QFile("/dev/fb0").exists()` check that gui v1 executes inside the device.

@jepefe @jhofstee what should we do? Close as won't fix or expose the display present value for gui-v2? This task is part of effort to bring gui-v2 up to par with gui-v1 changes.

kwindrem, 2024-06-03:
Whatever mechanism is used, keep in mind other platforms (eg Raspberry Pis) will have touch screens as well. Also some installs run a Cerbo with a third-party HDMI touch screen.

jpetrell, 2024-06-04:
Discussed in dev meeting that there is no reason to limit the GUI switch setting on gui-v2. Closing.

closed 2024-06-04 by jpetrell