victronenergy/gui-v2 #1098: gui-v2 doesn't have a settings page for "Modbus TCP/UDP devices"

DanielMcInnes, 2024-05-19:
TODO - check whether this makes sense

DanielMcInnes, 2024-05-19:
@ReinvdZee is this needed for gui-v2?

DanielMcInnes, 2024-05-19:
see gui-v1 commits:
b5c29b28a1792a9c242c20e40e641fbaae98034a
e0fee014090ad6ae0d44730c638dd7fca9e75bc6
6d3d1ce46dfee3241942dfd9de598cf40a7af3a3

ReinvdZee, 2024-05-21:
@DanielMcInnes,

The settings page you're referring to is this one: 
[image]

AFAIK, we need that page in Gui-v2 as well. I can't think of any reason why we would leave it out. 

@mpvader, can you confirm?

mpvader, 2024-05-21:
Yes; but I expect that there is something for that in the settings proposal project by Joona & Serj & Philipp. I’ll know more about that on Friday.

for now this one is blocked, to prevent double work - am I right @jpetrell ?

jpetrell, 2024-05-31:
> for now this one is blocked, to prevent double work - am I right @jpetrell ?

We have been discussing where Modbus device settings should go, but before we start the implementation effort there would initially just align the structure with gui-v1 for now.

mpvader, 2024-05-31:
👍

ReinvdZee, 2024-06-18:
Tested on physical system with a modbus TCP device discoverable. 

Automatic and manual scanning works. Device shows up under "Discovered devices". Device can be enabled/disabled from the "Discovered devices" menu, and shows up in the device list accordingly.  

**Bug**
The IP address validation does not work as expected. I get "Invalid IP address" on almost every IP address I enter. Earlier we established that this works fine on a mock system (@DanielMcInnes)

[Screencast from 18-06-24 12:29:19.webm](https://github.com/victronenergy/gui-v2/assets/38864570/2f102651-efdb-4eba-8801-17d879dfdfce)

DanielMcInnes, 2024-06-18:
@ReinvdZee , I think you have to press the 'Enter' key when you add an IP address before moving on to the next field, do you remember if you did that? Otherwise you will get 'Invalid IP address' for an IP address that appears to be fine

ReinvdZee, 2024-06-18:
@DanielMcInnes I did not press the enter key after entering the IP address. If I do, then indeed it works. 

Can we make it so the user does not have to press enter first? Because the IP address is usually the only field you'll have to fill. So I would expect it to work if I just press "Add device" after entering the IP address without pressing 'enter' first.

DanielMcInnes, 2024-06-18:
Yes, I found this irritating as well. I will edit the title for this ticket

jpetrell, 2024-06-18:
> Yes, I found this irritating as well. I will edit the title for this ticket

Also see earlier task #1047 Clearly indicate text field confirmation step.

blammit, 2024-06-18:
@DanielMcInnes This task hasn't changed; all the commits that refer to it are for the original issue of adding this settings page. Could you leave it with the original title? The text field acceptance issue will be fixed separately by Joona's PR.

DanielMcInnes, 2024-06-19:
Sure, changed it back

DanielMcInnes, 2024-06-27:
Closing this issue, as the settings page for Modbus TCP/UDP devices has been created, and the text field acceptance issue will be fixed separately by Joona's PR

closed 2024-06-27 by DanielMcInnes

DanielMcInnes, 2024-07-04:
Verified with Rein

commit e6ff5a6e (2024-06-10): Add 'Modbus TCP/UDP devices' page

Fixes #1098
commit f04c69b6 (2024-06-10): Add 'Modbus TCP/UDP devices' page

Fixes #1098
commit d7d43641 (2024-06-10): Add 'Modbus TCP/UDP devices' page

Fixes #1098
commit d4dd59cf (2024-06-10): Add 'Modbus TCP/UDP devices' page

Fixes #1098