victronenergy/gui-v2 #1101: PageSettingsHub4 should have a 'Dynamic ESS' item

DanielMcInnes, 2024-05-20:
See gui-v1 PageSettingsHub4.qml

blammit, 2024-05-30:
Fixed by https://github.com/victronenergy/gui-v2/pull/1155

chriadam, 2024-06-20:
As per testing on Tuesday:
1) If "Grid Metering" is not "External meter" then the Peak Shaving "Limit system AC import current" should have a popup saying why it cannot be enabled (i.e. because "Grid metering" is not set to "External meter").
2) "Scheduled charging" should be "Scheduled charge levels" and it should have a summary text in the list item "Inactive" if none exist.

(Question: after one schedule is enabled but not active, should the test read "Scheduled" rather than "Inactive"?)

blammit, 2024-06-30:
> If "Grid Metering" is not "External meter" then the Peak Shaving "Limit system AC import current" should have a popup saying why it cannot be enabled (i.e. because "Grid metering" is not set to "External meter").

@ReinvdZee instead of showing this popup, I've added this information below the switch when the switch is disabled. I think this provides the information more effectively as you don't have to click it in order to see the popup message and understand why it's disabled. Is this okay?

[image]

ReinvdZee, 2024-07-01:
@blammit, 

> @ReinvdZee instead of showing this popup, I've added this information below the switch when the switch is disabled. I think this provides the information more effectively as you don't have to click it in order to see the popup message and understand why it's disabled. Is this okay?

I agree, looking good!

blammit, 2024-07-02:
Changes have been merged; closing.

closed 2024-07-02 by blammit

DanielMcInnes, 2024-07-04:
verified with Rein

commit e53e9671 (2024-05-28): ess: add "Dynamic ESS" settings page

Fixes #1101
commit 3f70dffd (2024-05-30): ess: add "Dynamic ESS" settings page

Fixes #1101