victronenergy/gui-v2 #1104: Make spinbox number editable

blammit, 2024-05-20:
Make the number spinbox editable -- e.g. replace it with a text input when clicked -- so that the user has an alternative to just pressing the -/+ buttons, which may take a long time if the desired number is not close to the original value.

Some issues for investigation/discussion:

- Should the text input have a frame/border to indicate it is editable?
- How to show the VKB above the NumberSelectorDialog (possibly need to use a Popup instead of a Dialog)
- The VKB in the current position would cover the text input. Should the dialog/popup move upwards when it receives focus? Or what can be done to make the input fully visible?

jpetrell, 2024-05-20:
Also reported by the [customer](https://community.victronenergy.com/questions/279997/venus-os-v3408-available-for-testing.html?childToView=286325#answer-286325) independently "AC input current limit shows fractions, but increase/decrease works only +-1A, can't activate keyboard, can't set 16.2 or 16.0".

blammit, 2024-06-10:
The AC input current limit issue is fixed by https://github.com/victronenergy/gui-v2/pull/1199

MikeTrahearn-Qinetic, 2024-11-12:
The VKB overarching issue probably needs to be discussed first before any changes are made to the spin box with respect its primaryLabel and secondaryLabel in its contentItem.

MikeTrahearn-Qinetic, 2024-11-20:
This is also requested here:
https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1731734054739599

MikeTrahearn-Qinetic, 2024-11-20:
For reference: https://bugreports.qt.io/browse/QTBUG-128406

commit b8d35cea (2024-12-09): Make NumberSelectorDialog's SpinBox editable

Fixes #1104
commit 57e77145 (2024-12-11): Make NumberSelectorDialog's SpinBox editable

Fixes #1104
commit ad51a9a2 (2024-12-23): Make SpinBox editable

Upgrade the contentItem to use TextInput with correct VKB interaction.

Fixes #1104
commit bcac6598 (2024-12-23): Make SpinBox editable

Upgrade the contentItem to use TextInput with correct VKB interaction.

Fixes #1104
closed 2025-01-08 by MikeTrahearn-Qinetic