victronenergy/gui-v2 #1106: digital inputs: Add type for touch input control

DanielMcInnes, 2024-05-20:
See gui-v1 commit 0e3c23a0e2ad6b7d58280cf66d7d8f7af5269173

ReinvdZee, 2024-06-18:
Tested on Cerbo Mk2 with Venus OS V3.40~29

This does not work properly on V2 yet. Expected behavior:
- Set digital input type to "Touch input control"
- Toggle digital input, touch should be disabled and this toast should show up:
[image]
- When touching the display then another toast should show up notifying the user that the touch is disabled
[image]
- When toggling the input again, this toast should show up
[image]

In gui-v2, no toasts show up and the touchscreen always works. Reopening issue

kwindrem, 2024-06-18:
I would say navigation should not be disabled in the view only mode. Also, disabling changes on one display should not disable it on another (local vs multiple remote instances of gui-v2).

mpvader, 2024-06-19:
Lets discuss next week

blammit, 2024-06-27:
Discussed that:

- setting can be changed from wasm and device
- touch input disabling should only have effect on device

Additional features should be delayed to v1.1.0.

ReinvdZee, 2024-06-27:
Enabling / Disabling touch from gui-v2 is actually not as straightforward as we thought. 

`venus-platform` does create the path `Settings/Gui/TouchEnabled`, but does not handle the touch controller. There's a path just to store the setting. Gui-v1 controls the touch controller directly to enable and disable touch control. 

I don't think we should migrate this to venus-platform since a mouse (or touch) event is checked by the gui and rejected if touch is disabled. If we move this part to venus-platform, it will cause a lot of dbus traffic and decrease the responsiveness of the UI. V2 can probably implement it similar to V1. I'm not sure how that works on the WASM build, but thats for later. 

@jhofstee, @mpvader

DanielMcInnes, 2024-07-02:
@ReinvdZee , https://www.victronenergy.com/media/pg/Cerbo_GX/en/digital-inputs.html lists the different ways that a digital input may be configured. I notice that 'touch input control' is not listed there, should that be updated? I don't know who looks after this. It also suggests that if a digital input is configured as a touch input control, a new entry in the device list should appear for the touch input control. This doesn't happen on gui-v1, I'm not sure if it should or not, what do you think?

ReinvdZee, 2024-07-02:
Hi @DanielMcInnes, thanks for the heads-up. I agree that the documentation should be updated.
As for the device list, no device shows up when it is configured as touch input control because no service is created when the digital input is configured as touch input control because the gui directly handles it.

commit c01c77b3 (2024-07-03): Add type for touch input control

When a digital input is configured as 'Touch input control', and that digital
input is 0, gui-v2 should ignore touch events if running on the cerbo. Other
build types such as wasm should be unaffected, and handle touch inputs as
usual.

Show toast notifications on all build types when the touch input is enabled or
disabled.

Cerbo builds should show a notification 'Touch input disabled' when the screen
is touched while disabled.

Fixes #1106
commit 2a03e8f7 (2024-07-08): Add type for touch input control

When a digital input is configured as 'Touch input control', and that digital
input is 0, gui-v2 should ignore touch events if running on the cerbo. Other
build types such as wasm should be unaffected, and handle touch inputs as
usual.

Show toast notifications on all build types when the touch input is enabled or
disabled.

Cerbo builds should show a notification 'Touch input disabled' when the screen
is touched while disabled.

Fixes #1106
closed 2024-07-08 by DanielMcInnes