victronenergy/gui-v2 #1108: Numbers are inconsistently formatted in non-en locales

blammit, 2024-05-20:
gui-v2 should format numbers consistently in non-en locales.

For example, in Chrome if you set the "Preferred language" to German, then when running gui-v2 in wasm, then the Brief and Overview pages show quantities with the '.' period as a decimal separator:

[image]

However, when opening an Inverter/Charger control card, the current limit spinbox uses a comma (`,`) as the decimal separator:

[image]

gui-v2 should just use one or the other instead of mixing them.

blammit, 2024-05-20:
This is causing a bug in the "Minimum and maximum gauge range settings" as reported in https://github.com/victronenergy/gui-v2/pull/1043#issuecomment-2114044833. In that issue, if the locale is de_DE, the page shows the number with a period as a decimal separator, but requires that the input uses a comma as a decimal separator.

blammit, 2024-05-21:
Part of the problem is that in wasm, there are multiple locale settings:

- the locale set within gui-v2 based on the language settings
- the default locale, as set by the browser

If these differ, then gui-v2 needs to pick one of these and follow it consistently.

mr-manuel, 2024-05-22:
I tested this build and it seems that it gets the locale of the browser and not of the language setting of Venus OS.

This build:
[image]

Main branch:
[image]

Additionally there is still a strange behaviour:
[image]

mr-manuel, 2024-05-22:
Now it looks good!

## Venus OS in German
[image]
[image]

## Venus OS in English
[image]
[image]

Remains only the issue, if you enter `.` instead of `,` (or vice versa, depending on the language) you get an odd answer. I don't know, if you want to fix that also. If yes, I don't see it that critical for `v1.0.0` release.

blammit, 2024-05-27:
> Remains only the issue, if you enter . instead of , (or vice versa, depending on the language) you get an odd answer. I don't know, if you want to fix that also. If yes, I don't see it that critical for v1.0.0 release.

Thanks for verifying! Yes, this issue does remain, but we will leave it for now.

closed 2024-05-28 by blammit

blammit, 2024-12-10:
Note: the resolution here was to always use the C locale to format numbers, as users may prefer the dot as a decimal separator, regardless of the application locale.

commit 6a0e9887 (2024-05-21): Consistently format numbers according to system locale

In QML, use toLocaleString() instead of toFixed() to format numbers.
The former respects the decimal separator for the locale, while the
latter always uses a period as the decimal separator.

In C++, use QLocale::toString() to use the correct locale for number
formatting.

Fixes #1108
commit 00afb851 (2024-05-21): Consistently format numbers according to application locale

In QML, use toLocaleString() instead of toFixed() to format numbers.
The former respects the decimal separator for the locale, while the
latter always uses a period as the decimal separator.

In C++, use QLocale::toString() to use the correct locale for number
formatting.

Fixes #1108
commit f5fe6e2d (2024-05-21): Set application locale and use locale-specific number formatting

Call QLocale::setDefault() when the current language is applied. This
allows Qt.locale() to return the locale for the current language, and
also this locale will be used as the default locale, e.g. when
formatting date/times and floating point numbers for spinboxes and
text fields.

In QML, always use toLocaleString() instead of toFixed() to format
numbers. The former respects the decimal separator for the locale,
while the latter always uses a period as the decimal separator.

In C++, use QLocale::toString() to use the correct  [... cut]
commit 0875c6b9 (2024-05-22): Always format numbers according to the locale of the app language

Ensure the locale of the language selected in the application settings
is the one used to format all numbers in the application, rather than
the locale of the system (or the browser, when loading in wasm).

Calling QLocale::setDefault() is not sufficient to ensure that the
correct locale is loaded by Qt.locale(), so pass the locale name to
Qt.locale() to ensure the correct locale is used when formatting
numbers and date/times.

Also, always use toLocaleString() instead of toFixed() to format
numbers. The former respects the dec [... cut]