victronenergy/gui-v2 #1118: Quantity table missing on the Solar charger settings

jpetrell, 2024-05-21:
[Reported](https://community.victronenergy.com/questions/279997/venus-os-v3408-available-for-testing.html) by the customer, "With 3.40~10, details of solar (MQTT Solar) are not seen any more"

[image]

blammit, 2024-05-22:
I'm having trouble reproducing this one; have tried several different installations on VRM. It appears that the QuantityTableSummary is producing invalid data, and therefore it has no children and the height is shrunk, but it's strange that even the default text (e.g. "--" if the voltage is not present) is not appearing.

jpetrell, 2024-05-24:
> I'm having trouble reproducing this one

Reducing priority. I can check at some point too, if cannot repro either will close.

jpetrell, 2024-06-03:
No reply from the customer yet to the comments.

blammit, 2024-06-28:
Can reproduce this on the system reported by https://github.com/victronenergy/gui-v2/issues/1280. Closing this as duplicate.

closed 2024-06-28 by blammit