victronenergy/gui-v2 #1125: Consider turning of the side panel graphs on high CPU usage

jpetrell, 2024-05-22:
After #914 gui-v2 started pausing animations on the Overview page if accumulative CPU load of all the CPUs goes above 85%. Animations become secondary during heavy load (e.g. complex installations with low-powered Cerbo GX) and the system resources need to be freed for more critical background processes.

Consider doing similar thing for the Brief page side panels that can cause fair bit of CPU usage also, especially now with up to 4 graphs animated simulatenously. Tested that I was getting 39-42% CPU usage on Cerbo with 3 animating graphs, and possibly even more with 4 graphs (didn't test).

Though cannot just pause the graphs Pas then the system will start to show invalid, stale data.

jpetrell, 2024-05-22:
This feels equivalent to the Overview page case #914 so assigning the same medium priority.

jpetrell, 2024-05-23:
Discussed with @mpvader there are more important items to tackle right now, reducing priority.

mpvader, 2024-05-27:
the background: in case of issues, whatever issues, including cpu usage, we'll remove the side panel entirely for v1.0.0.

jpetrell, 2024-05-31:
@jepefe what do you think? Would the above solution be acceptable?

jpetrell, 2024-06-03:
There is a risk that the user finds these banners annoying, but the closing may appear as bug if we are not telling the user. We should do something, if the system load is very high the background processes take priority to drawing side panel graphs.

Discussed with @mpvader that let's put the PR #1174 in to get feedback.

PR #1174 (2024-06-03): Close side panel when detecting high CPU usage
Solution proposal: automatically collapse the side panel if CPU utilization goes over 90% and display banner "System load high, closing the side panel to reduce CPU load".

Fixes #1125.
commit ae14eeb3 (2024-06-03): Close side panel when detecting high CPU usage

Fixes #1125.
closed 2024-06-03 by jpetrell