victronenergy/gui-v2 #1137: VeBus and InverterCharger settings expose same device details, but in slightly different order

jpetrell, 2024-05-23:
Actual result: Seems two different DBus services have been implemented separately, causing small differences in the way the information is presented.

Expected result: Device details should be presented in consistent manner. Ultimately both display the same information.

[image]

blammit, 2024-05-24:
Yeah, this is a bit odd. This is happening because `OverviewInverterChargerPage` comes from the Figma design for the vebus drilldowns, while `PageVeBus.qml` is the ported version from gui-v1. 

Perhaps the Settings 2.0 will lead to an updated design for `PageVeBus.qml`?

blammit, 2024-08-09:
Discussed this with @jepefe . The overview drilldown is supposed to show generic information that is relevant to VE.Bus devices, RS systems and inverters, whereas `PageVebus.qml` only provides VE.Bus information.

commit 93e7d427 (2024-08-16): Use OverviewInverterChargerPage for inverter, vebus and acsystem services

The overview drilldown is supposed to show generic information that is
relevant to VE.Bus services, RS systems and inverters, instead of only
showing data for VE.Bus services.

Fixes #1137
commit 510130b6 (2024-08-16): Use OverviewInverterChargerPage for inverter, vebus and acsystem services

The overview drilldown is supposed to show generic information that is
relevant to VE.Bus services, RS systems and inverters, instead of only
showing data for VE.Bus services.

In PageRsSystem, show the current limits for all AC inputs, to align
with OverviewInverterChargerPage.

Fixes #1137
commit b170224e (2024-08-19): Use OverviewInverterChargerPage for inverter, vebus and acsystem services

The overview drilldown is supposed to show generic information that is
relevant to VE.Bus services, RS systems and inverters, instead of only
showing data for VE.Bus services.

In PageRsSystem, show the current limits for all AC inputs, to align
with OverviewInverterChargerPage.

Fixes #1137
commit a8e8449e (2024-08-19): Use OverviewInverterChargerPage for inverter, vebus and acsystem services

The overview drilldown is supposed to show generic information that is
relevant to VE.Bus services, RS systems and inverters, instead of only
showing data for VE.Bus services.

In PageRsSystem, show the current limits for all AC inputs, to align
with OverviewInverterChargerPage.

Fixes #1137
closed 2024-08-19 by blammit