victronenergy/gui-v2 #1142: Multi RS solar device duplicated in Device List and not listed in "RS devices"

blammit, 2024-05-27:
Tested with Groningen MultiRS test installation. 

The MultiRS solar device is duplicated in the Device List, and also is not listed as part of a "RS devices" settings list item in the solar device settings, as per gui-v1.

The "RS devices" will be addressed by #1077, but verify that the Device List duplication is fixed as well.

podarok, 2024-05-27:
My MultiRS is listed in RS Devices
[image]

[image]

Venus OS version
v3.40~17
[image]

But it is duplicated in gui-v2, which is an issue
[image]

blammit, 2024-06-05:
PR merged, closing.

blammit, 2024-06-05:
@podarok Thanks for the screenshots -- the Device List should now be fixed.

closed 2024-06-05 by blammit

podarok, 2024-06-05:
> @podarok Thanks for the screenshots -- the Device List should now be fixed.

Is there a new Venus OS beta build where I can check this?

jpetrell, 2024-06-05:
> Is there a new Venus OS beta build where I can check this? @blammit

We will tag a new gui-v2 release end of this week, after which it will be picked for the next [Venus OS beta](https://community.victronenergy.com/questions/279997/venus-os-v3408-available-for-testing.html) release in few days.

blammit, 2024-06-06:
@podarok this change hasn't been released yet -- hopefully the next one!