victronenergy/gui-v2 #1144: Wrong gauge values when some phases consume while others feed in

blammit, 2024-05-27:
When consuming on one phase and feeding in on another, the gauge values for the feed-in phases are zero.

[image]

blammit, 2024-06-06:
Would be nice to have a test system for this, as I cannot reproduce in mock mode.

@ReinvdZee would it be possible to have a python script with a recording to test this situation, where some phases are negative while others are positive?

DanielMcInnes, 2024-06-17:
I can reproduce this issue at "GRN - RS 3phase without EM". It is caused by clamping the power value when grid feedback is disabled. I removed the clamping, and gui-v2 showed the same gauge values as gui-v1. I'll check for clamping elsewhere in the codebase

chriadam, 2024-06-18:
If grid feedback is disabled, why should any negative power values be seen?

chriadam, 2024-06-18:
Or, to ask the question a different way: is the system not disabling grid feed-in properly, in the case where the user specifies that grid feed-in should be disabled?

blammit, 2024-06-19:
@jepefe we're seeing an installation (RS 3phase without EM) with these settings:

`/Hub4` is not set
`/Settings/CGwacs/OvervoltageFeedIn` = 0
`/Settings/CGwacs/PreventFeedback` = 0

This should mean that feed-in is not enabled. As mentioned [here](https://github.com/victronenergy/gui-v2/issues/1041#issuecomment-2106903897), this is the test, for whether feed-in is enabled:

```
com.victronenergy.system/Hub == 4 && 
(com.victronenergy.settings/Settings/CGwacs/OvervoltageFeedIn == 1 || 
com.victronenergy.settings/Settings/CGwacs/PreventFeedback == 0)
```

So on the installation I would expect power/current values to always be positive (or can be ignored if negative). However, the installation is producing negative values, and VRM portal is showing those as feed-in values.

What should gui-v2 be doing here? Should the feed-in conditional be modified?

DanielMcInnes, 2024-06-19:
@chriadam , possibly the system is not disabling grid feed-in properly, but that is another issue. We should still be displaying the 'real' values showing grid feedback, and matching gui-v1's behaviour.

chriadam, 2024-06-19:
That's true, I guess, if something wrong is happening it's better to SHOW the user that something is wrong.

blammit, 2024-06-19:
The checking of whether feed-in is disabled is something new introduced in gui-v2, so perhaps it's an unreliable check at this stage. And in the interests of aligning with VRM Portal and gui-v1 I think it'd be better to drop it for now.

I think really, the values should be clamped in the backend, and not at the UI level.

jepefe, 2024-06-20:
The problem is that those paths works for VE.Bus Inverters but probably not for RS inverters.

@izak Can you confirm which paths shall we look into to determine if ESS is enabled for Multi RS? Is there a generic way to check it for both, RS & VE.Bus?

izak, 2024-06-27:
@jepefe Technically ESS is always "enabled" for the Multi-RS. There is nothing special you have to do, it always has the capability. There is [an issue][1] for providing a single place that will tell you if feed-in is expected or not.

Overall, I would agree with the earlier statement that hiding feed-in from the user (aka clamping) should not be done. In a three phase system that balances the total of the phases to a positive value, it is still possible to have some of your phases negative.

[1]: https://github.com/victronenergy/venus-private/issues/370

commit eccd66d1 (2024-06-18): Wrong gauge values when some phases consume while others feed in

Fixes #1144
commit 8616099b (2024-06-18): Wrong gauge values when some phases consume while others feed in

Fixes #1144
commit 9a85cea0 (2024-07-03): Wrong gauge values when some phases consume while others feed in

Fixes #1144
commit 0ebc8ceb (2024-07-03): Wrong gauge values when some phases consume while others feed in

Fixes #1144
closed 2024-07-03 by DanielMcInnes