victronenergy/gui-v2 #1145: Invert gauge bars when feed-in is enabled

blammit, 2024-05-27:
[image]

The confusion arises because when feed-in is enabled, the power/current is negative, so the gauge bars move in the opposite direction. We could use an absolute value for the gauge instead when feed-in is enabled, so that a longer green bar indicates more power being fed back to the grid, rather than less.

mr-manuel, 2024-05-27:
Ask the user to switch to amps display instead of watts to see the difference. Could be the same as https://github.com/victronenergy/gui-v2/issues/1045.

mr-manuel, 2024-05-27:
Please do not change it. I think that would be even more confusing. Let's try to solve the other issue first and see if it also solves this one.

mpvader, 2024-05-27:
Hey @mr-manuel , how is making the light green section of the bar longest for the phase that feeds most power in not good?

i see how the other issue can make things extra confusing, but still it looks like there is a small bug here?

mr-manuel, 2024-05-27:
I meant that it could be confusing if you only change colors and use absolute values. At least it would confuse me. A nice thing would be, if the bar is starting at 0 and not at minimum. Therefore it should go down or up.

mr-manuel, 2024-05-27:
I re-read everything and maybe I misunderstood. I think this would be the clearest option. Also for the gauges on the overview page, if feed in is enabled.

[image]

[image]

jepefe, 2024-05-28:
The idea is to use absolute AC IN values for the gauge and change the color of the bar to green when values are negative which means feeding back the grid. 

The advantages are:
- Easy to read from the distance as it uses the whole gauge
- Don't need to learn to read a split or a bidirectional gauge
- Keeps UI coherence
- Simple implementation

kwindrem, 2024-05-29:
I agree with mr-manual. The color difference is not sufficient to differentiate consumption vs feed-in to an unskilled user. So please revisit this after gui-v2 v1.0 is released. Thanks.

commit 1ad62512 (2024-06-06): Invert arc and bar gauges when feeding in to grid

When feeding power to the grid, use an absolute value for the side
arc gauges and bar gauges. This effectively reverses the gauge
direction so that negative and positive values have the same value on
the gauge, though negative values will be drawn in green.

Fixes #1145
commit b371e692 (2024-06-07): Invert arc and bar gauges when feeding in to grid

When feeding power to the grid, use an absolute value for the side
arc gauges and bar gauges. This effectively reverses the gauge
direction so that negative and positive values have the same value on
the gauge, though negative values will be drawn in green.

Also ensure that when feed-in is enabled, we use the (abs value)
max of the minimum and maximum range values as the maximum range,
and zero as the minimum range.

Fixes #1145
PR #1191 (2024-06-07): Invert arc and bar gauges when feeding in to grid
When feeding power to the grid, use an absolute value for the side arc gauges and bar gauges. This effectively reverses the gauge direction so that negative and positive values have the same value on the gauge, though negative values will be drawn in green.

Fixes #1145
commit f99e0146 (2024-06-07): Invert arc and bar gauges when feeding in to grid

When feeding power to the grid, use an absolute value for the side
arc gauges and bar gauges. This effectively reverses the gauge
direction so that negative and positive values have the same value on
the gauge, though negative values will be drawn in green.

Also ensure that when feed-in is enabled, we use the (abs value)
max of the minimum and maximum range values as the maximum range,
and zero as the minimum range.

Fixes #1145
closed 2024-06-07 by chriadam