victronenergy/gui-v2 #1150: Virtual keyboard doesn't open when trying to type text on gui-v2 on the browser

jpetrell, 2024-05-27:
Open gui-v2 on a mobile device from VRM website. Tap any text field.

Expected result: Virtual keyboard opens, allowing you to input text.

Actual result: Blinking cursor becomes visible, indicating the text field thinks it has focus, but no keyboard appears.

DanielMcInnes, 2024-05-28:
If I connect to a cerbo directly via my phone, eg. browse to https://10.0.0.2/gui-beta/ , I can enter text ok.
[image]

If I connect to a cerbo via VRM on my phone, I see the issue, and when I click in a text entry field, no keyboard pops up.

jpetrell, 2024-05-28:
Does not repro on iPad (and likely then also not on iPhone either), only on Android. This may be something VRM adds around the gui-v2 that Qt doesn't like, or a bug in Qt itself. Reading the source codes (commits [here](https://codereview.qt-project.org/c/qt/qtbase/+/260717) and [here](https://codereview.qt-project.org/c/qt/qtbase/+/295032)) looks like Qt creates an off-screen, hidden <input type=“text” contenteditable=true> element and gives it focus when a Qt input gains focus.

jpetrell, 2024-05-31:
Fixed in the upcoming VRM website release, "the popup used for the remote console does not allow focusing on elements outside of it, therefore making the proxy input inaccessible".

closed 2024-05-31 by jpetrell