victronenergy/gui-v2 #1154: After typing the virtual keyboard always comes up on a tap on Android (wasm)

jpetrell, 2024-05-28:
Open gui-v2 on the local network by pointing Android browser to [your_device_ip_address]/gui-beta. Tap any text field to open virtual keyboard, e.g. one of the fields in Settings > Display & Languages > Minimum and maximum gauge ranges. Virtual keyboard should open.

Now close the keyboard and try to perform some other task that doesn't involve typing text.

Actual result: After opening the keyboard every time you touch the gui-v2 the keyboard opens again.

Debugging with Chrome remote device debugging `chrome://inspect/#devices` I can see the `<body>` being the document.activeElement initially. After tapping the field the focus becomes `<input>` element, which it stays even if you tap outside the field and/or closing the keyboard.

https://github.com/victronenergy/gui-v2/assets/2203667/dd10ebac-08d1-40e9-ab97-7c172a2958a5

jpetrell, 2024-05-28:
There are old [Qt issues](https://bugreports.qt.io/browse/QTBUG-88803) about Android keyboard always popping up, but such items are closed. Upgrading Qt might solve the issue, or we could work-around the issue by trying to set the `<input type=“text” contenteditable=true>` element contenteditable value to false whenever text field loses active focus inside the Qt window.

chriadam, 2024-05-29:
Looks like the bug was closed by Qt even though the issue was never fixed.  I have pinged.

There might be some issue with Android Chromium also: https://issues.chromium.org/issues/40912131

commit f8bf2bd9 (2024-06-03): Suppress Android keyboard if no text field has focus

Fixes #1154.
commit 9769e264 (2024-06-04): Suppress Android keyboard if no text field has focus

Fixes #1154.
PR #1182 (2024-06-04): Suppress Android keyboard if no text field has focus
Fixes #1154.
commit 064131fb (2024-06-04): Suppress Android keyboard if no text field has focus

Fixes #1154.
closed 2024-06-04 by jpetrell