victronenergy/gui-v2 #1166: Input field feedback: read-only not understandable

mr-manuel, 2024-05-30:
For most items it's not clear if they are enabled/disabled or readonly.

All fields are changeable now:

[image]

All fields are read-only now:

[image]

The only difference is, that an enabled button gets grayed.

jpetrell, 2024-06-03:
With fixes made for #934 (that were just merged) the read-only text fields are now displayed as regular read-only text items:

[image]

blammit, 2024-06-03:
Closing as fixed.

closed 2024-06-03 by blammit