victronenergy/gui-v2 #1178: Victron Energy Meters: add option to configure the location and phase setting

mpvader, 2024-06-01:
This is about the settings in below screenshot.

A) For all previous energy meters, those settings were in the GX menus (gui-v1)

B) For all other products, except GX, we have all settings via VictronConnect.

For our own energy meter, VM-3P75CT, we decided to have it in VictronConnect, for consistency with (B). But now for anyone familiar with the GX & enery meters, ie electrical installers doing ESS systems, its hard to find the setting.

I see two a few options to solve that:

1. add these settings to gui-v2; 
2. add a placeholder text saying to find them in VictronConnect
3. at least show in gui-v2 what the settings are set to, so its not as obscure as now.

here are the settings:
[image]

and they are already on dbus, so thats easy:
[image]

blammit, 2024-06-03:
gui-v2 doesn't have a settings page for "Energy Meters" at the moment, but is this in regards to the settings page for PV inverters? The PV Inverter overview drilldown currently looks like this:

[image]

This already shows the "Position", so we could add the "Role" as well. What does "Phases Number" refer to? Is that related to`NrOfPhases` or something else?

ReinvdZee, 2026-09-10:
Discussed in the meeting that @blammit will test this.