victronenergy/gui-v2 #1185: Ensure that gui-v2 VRM connections don't subscribe to wildcard topics

chriadam, 2024-06-04:
VRM-launched (or connected) gui-v2 instances should know their portal id, and should never subscribe to wildcard topics.

Double check that this is the case.

chriadam, 2024-06-05:
My code which detected whether we are connected to VRM or not was too fragile, didn't accept query parameters.
https://github.com/victronenergy/gui-v2/pull/1189 fixes this.

mpvader, 2024-06-08:
Hey @chriadam , I think the change here https://github.com/victronenergy/veutil/commit/0eeb71b24936f7d37730bb51da2f62b25fa2f37b broke gui-v2 on VRM.

And, without being sure that I understand the whole thing, it looks to me like this can be done better in general. Instead of waiting for the vrm portal id to come in via MQTT, you can take it from the query parameters.

And if it would help to do the same thing when running locally, we can probably, and with a bit of effort, do the same there I'd think.

chriadam, 2024-06-10:
We do take it from query parameters, and also from arguments if provided.  https://github.com/victronenergy/gui-v2/blob/main/src/main.cpp#L74

We only parse it from MQTT if no information is given on query/argument (such as when connecting to the local LAN, where we don't know the portal id).

chriadam, 2024-06-10:
But there was a bug - it wasn't passed to the backend prior to opening connection.  Fix PR up now.

commit bbadf492 (2024-06-10): MQTT: ensure the portal id parameter is set before connecting

Previously, it was set after the connection was initiated, but
before the connection was established (i.e. at the same time the
username and password were provided).

Now, it is provided at the very start of the flow, so that the
producer can emit an error if no portal id is provided for
VRM connections.

Also, align the detection of VRM addresses between gui-v2 and
veutil code.

Fixes #1185
PR #1207 (2024-06-11): MQTT: ensure the portal id parameter is set before connecting
Previously, it was set after the connection was initiated, but
before the connection was established (i.e. at the same time the
username and password were provided).

Now, it is provided at the very start of the flow, so that the
producer can emit an error if no portal id is provided for
VRM connections.

Also, align the detection of VRM addresses between gui-v2 and
veutil code.

Fixes #1185
commit 39b30e6e (2024-06-11): MQTT: ensure the portal id parameter is set before connecting

Previously, it was set after the connection was initiated, but
before the connection was established (i.e. at the same time the
username and password were provided).

Now, it is provided at the very start of the flow, so that the
producer can emit an error if no portal id is provided for
VRM connections.

Also, align the detection of VRM addresses between gui-v2 and
veutil code.

Fixes #1185
closed 2024-06-11 by chriadam