victronenergy/gui-v2 #1193: Auto selected BMS displaying empty value

jpetrell, 2024-06-06:
The automatically selected BMS is not displayed in GUI-v2, while it is displayed correctly in GUI-v1. Tested 3.40~21 GUI-v2. Reported on venus-tests. 

[image]

blammit, 2024-06-07:
On this system, the `/ActiveBmsService` and `/ActiveBmsInstance` are
valid, but the MQTT service referenced by this combination of service name and instance does not exist. It does exist on dbus, though. The issue could be related to the system using a virtual BMS.

Will check with @ReinvdZee on this one.

jpetrell, 2024-06-12:
Doesn't seem common issue, lowering priority.

ReinvdZee, 2024-06-12:
See [this](https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1717726127695089) slack thread on the missing MQTT service. This will be merged in a coming release.

blammit, 2024-06-12:
Closing as this will be fixed in VenusOS.

closed 2024-06-12 by blammit