victronenergy/gui-v2 #1211: Order of AC input phase gauges reversed in the feed-in mode

jpetrell, 2024-06-10:
Reported by [customer](https://community.victronenergy.com/questions/279997/venus-os-v3408-available-for-testing.html): "The ratio of the gauges to each other looks better. But it is exactly the other way round. L1 is the smallest value with the longest gauge and vice versa."

[image]
[image]

jpetrell, 2024-06-10:
Checking with ESS demo using Ekrano GX I can see with dbus-spy that gui-v2 phase values and gauges match the values reported by com.victronenergy.system/Ac/ActiveIn/L1-3.

blammit, 2024-06-10:
Looks like this is a duplicate of https://github.com/victronenergy/gui-v2/issues/1145? The gauges in the screenshot look about right if using the implementation prior to the fixes for that task. Closing as duplicate.

closed 2024-06-10 by blammit

jpetrell, 2024-06-11:
> Looks like this is a duplicate 

You're right, my bad. 3.40~22 didn't have the absolute values change yet that was released in v3.40~26. The comment being recent through me off.