victronenergy/gui-v2 #1218: Add option to boot straight to the Overview page if preferred over the Brief page

jpetrell, 2024-06-10:
Requested by quite a few members of [community](https://community.victronenergy.com/questions/292058/venus-os-v34026-available-for-testing.html). Could be a user option or system could remember which page was visited the last. 

There is also similar issue "Allow making Environment tab default on the Levels page in case you are not that much interested in tanks" that maybe could be tackled at the same if a suitable pattern is found.

mpvader, 2024-06-10:
Hey, to add: Its an important one to me, mostly the choosing between Brief and Overview page.

Some users will like the brief page, some other personas / type of users will prefer the overview page instead. hence we need to preferably make it such that they don't have to do a click everytime that they open the GUI in Remote Console.  (which is when it starts from scratch).

something else that comes to mind there is the animation at start-up: is something being loaded while doing that animation? or is it just the animation because its nice? Same thing: first few times its very nice, but after a while its not very nice anymore - at least not to the more impatient people 😄 .

kwindrem, 2024-06-10:
Rather than an option, why not make it work like gui-v1 and return to the last viewed overview page?

mpvader, 2024-06-10:
Thats also what the post by Joona says :-)

mpvader, 2024-06-10:
I marked this high prio, bit not for v1.0.0

kwindrem, 2024-06-10:
Don't forget there can be multiple GUIs running simultaneously and each one could be on a different page so the page each GUI returns to may be different. Local storage on each platform running the GUI will be needed or you need to store each page to return to in dbus Settings with a unique identifier for each GUI.

blammit, 2024-06-11:
Related to https://github.com/victronenergy/gui-v2/issues/700.

`com.victronenergy.settings/Settings/Gui/DefaultOverview` controls the startup page in gui-v1.

jpetrell, 2024-06-11:
Discussed with Serj, he was cautious of adding any additional options or states, saying he would like to do more user testing before making such decisions and that currently you only just need one tab press after the launch to open the Overview and can then leave it on.

> something else that comes to mind there is the animation at start-up: is something being loaded while doing that animation? or is it just the animation because its nice?

Watching the animation takes multiple seconds, but also loading it before it can be displayed takes surprising amount of time, multiple seconds. I don't remember exact figures by heart, but skipping the animation step altogether loaded the gui-v2 in roughly half the time (just gui-v2, not counting the rest of Venus OS booting).

chriadam, 2024-09-04:
Following up on Joona's last comment here: that animation has been disabled for wasm builds, and yes it was a very significant time saving (since loading the animation takes a long time, and playing the animation takes a long time).

Nonetheless, my personal opinion is that allowing launch to specific page does actually make sense (for VRM especially).

chriadam, 2024-10-30:
@blammit can this be closed since https://github.com/victronenergy/gui-v2/pull/1591 was merged?

chriadam, 2024-10-30:
Issue looks like a duplicate of #1496

blammit, 2024-11-08:
Yep, let's close this.

closed 2024-11-08 by blammit