victronenergy/gui-v2 #1220: Connected gensets UI changes

ReinvdZee, 2024-06-11:
We've done quite some changes to the UI of connected gensets on gui-v1. See [this](https://docs.google.com/document/d/1NohCfWdM3Aj-FIk6D3GXrWsDlFFxs05h/edit?usp=drive_link&ouid=113747478174641618706&rtpof=true&sd=true) doc. These changes should be implemented in Gui-v2 as well.

The new v1 UI looks like this: 
[image]

Manual start menu: 
[image]

The v2 UI currently is: 
[image]

**Alarms**
`dbus_generator` now contains two seperate alarms for monitoring the auto start/stop mode of the GX and of the genset. Both alarms can be enabled/disabled with **one** switch (already present in Gui-v2):
[image]
The alarms are issued by `dbus_generator` and published by `venus-platform`. As was already discussed [here](https://github.com/victronenergy/gui-v2/issues/1086) no UI changes are necessary for the alarms to show up in Gui-v2.

**Other issues**
The genset we're testing with also exposes a tank sensor. On gui-v1, this is picked up correctly. Gui-v2 shows the sensor but it is "unknown":
[image]
On the tanks overview it is displayed properly: 
[image]

Tasks:

- [ ] Remove the "Mode" option
- [ ] Move the autostart switch and manual start menu to the top-level
- [ ] Place status and error code of the GX and the genset on the top-level
- [ ] Add "Remote start mode" status beneath the phase information
- [ ] Modify the manual start menu to disable the menu items when the genset's remote start mode is disabled. Also display this text then: "The remote start functionality is disabled on the genset. The GX will not be able to start or stop the genset now. Enable it on the genset control panel"

The menu is probably interleaved with the generator start/stop menu, be sure not to change the generator start/stop menu.

chriadam, 2024-06-12:
This seems like a large number of UI changes / feature changes.  There were other similarly large changes (like the acsystem one) recently, also.  I am surprised that we are still receiving large feature changes at this point, considering the timeline.  I would ask: is this truly necessary for 1.0.0?

blammit, 2024-06-13:
See [b9583fd7bee8f1120402ba4697693765c6f06f26](https://github.com/victronenergy/gui/commit/b9583fd7bee8f1120402ba4697693765c6f06f26) in gui-v1 for the main changes.

chriadam, 2024-06-20:
Reopening issue.  Per testing on Tuesday:
1) settings -> device list -> DSE genset controller, the text "Error" should read "Control error code" so as to be distinct from "Genset error code" which comes later in the list.
2) Genset device is not showing up on Overview page as an AC Input (nor as a gauge on Brief page).  Simulation N.

ReinvdZee, 2024-06-27:
Reported on slack: The generator manual start menu is not properly scaling. I reproduced this on a mobile phone. The controls work, but they do not scale evenly with the menu background. 

[image]

chriadam, 2024-07-04:
That latter issue (scaling of the dialog) is tracked separately by #1271

blammit, 2024-07-09:
For reference, use [this](c0619ab66479) installation to test the "DSE genset controller".

blammit, 2024-07-10:
> Genset device is not showing up on Overview page as an AC Input (nor as a gauge on Brief page). Simulation N.

This is because `com.victronenergy.system/Ac/ActiveIn/Source` is not set to a valid value, and gui-v2 relies on this to determine the current active AC input. (Also, `Ac/In/0/Connected` and `Ac/In/1/Connected` are not set.)

Normally `/Ac/ActiveIn/Source` is set, so I think this is a bug in the simulation. @ReinvdZee could you confirm whether this is the case?

ReinvdZee, 2024-07-11:
@blammit `/Ac/ActiveIn/Source` is only set when there is a multi with an active input. Simulation N is only a recording of the genset, not of the vebus service, so if you run this on a system with no further devices, `/Ac/ActiveIn/Source` will indeed not be set correctly. 

You can use simulation O to test this. Simulation O also has a vebus system. I ran it and the generator widget showed up:

[image]

I would expect it to show up on the top-left instead of the bottom-left, though. But that may be a design choice. 

We may not even need a genset service to test this. I set AC input 1 to "generator" on the [lynx parallel test system](https://vrm.victronenergy.com/installation/402872/dashboard) and it shows up:
[image]

blammit, 2024-07-16:
Note: I am currently clarifying with Rein how we should identify the active AC input, as it seems that /Ac/ActiveIn is only set when there is a multi device (like vebus) present. If this is the case, then some work is required to rearrange how the active AC input is determined.

blammit, 2024-07-19:
Clarified this with Rein. gui-v2 can use `/Ac/ActiveIn/Source` for now. I will close this as all issues are resolved.

commit 5e5161b6 (2024-06-14): UI changes for connected gensets

This ports b9583fd7bee8f1120402ba4697693765c6f06f26 and related
changes from gui-v1.

Fixes #1220
commit c54113cc (2024-06-14): UI changes for connected gensets

This ports b9583fd7bee8f1120402ba4697693765c6f06f26 and related
changes from gui-v1.

Fixes #1220
closed 2024-07-19 by blammit