victronenergy/gui-v2 #1224: List of batteries from overview drilldown is incomplete

blammit, 2024-06-12:
Clicking on the battery from the overview page shows a list of batteries in the system, if there is more than one battery.

The battery list is generated from `com.victronenergy.battery.*` but this is not sufficient. Some batteries are missing; see [this](https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1718001991024539) slack thread.

blammit, 2024-06-12:
Marking as blocked; need more info about where to find a complete list.

mpvader, 2024-06-14:
Hey @blammit, see showing all batteries chapter in UI document

(Still needs some input from Izak)

closed 2024-07-01 by blammit

blammit, 2024-07-12:
Note: the conclusion here was that the list of batteries should come from `com.victronenergy.system//Batteries`.

commit 0e065463 (2024-06-18): Fix incomplete battery list in Global.batteries.model

Show a complete list of batteries when the battery overview widget is
clicked. The list should come from com.victronenergy.system
/AvailableBatteries rather than com.victronenergy.battery.*.

Fixes #1224
commit 9e571117 (2024-06-18): Fix incomplete battery list in Global.batteries.model

Show a complete list of batteries when the battery overview widget is
clicked. The list should come from com.victronenergy.system
/AvailableBatteries rather than com.victronenergy.battery.*.

Fixes #1224
commit b675abb1 (2024-06-18): Fix incomplete battery list in Global.batteries.model

Show a complete list of batteries when the battery overview widget is
clicked. The list should come from com.victronenergy.system
/Batteries rather than com.victronenergy.battery.*.

Fixes #1224
commit fe375303 (2024-06-18): Fix incomplete battery list in Global.batteries.model

Show a complete list of batteries when the battery overview widget is
clicked. The list should come from com.victronenergy.system
/Batteries rather than com.victronenergy.battery.*.

Fixes #1224