victronenergy/gui-v2 #1232: Redesigned top-level structure for the settings

jpetrell, 2024-06-13:
GX devices have accumulated a lot of settings over time, the structure hasn’t kept up with all the changes. Finding what you are looking for can be difficult. Re-organize top-level settings in logical groups for easier access, e.g. Wi-Fi, Ethernet, GSM and Bluetooth under Connectivity. See the Figma design for more details.

DanielMcInnes, 2024-12-18:
https://www.figma.com/design/j9qNQjp1rHQSbJZRy0sxvk/GX-Touch---Implementation?node-id=6709-103273&p=f&t=YCK5nk5ZRgtQkTzi-0

DanielMcInnes, 2024-12-24:
Questions for @philipptrenz :
In the new design, is `System Setup -> Charge Control` the same as the old `Settings -> DVCC` ?

philipptrenz, 2024-12-27:
@DanielMcInnes it relates to
`System Setup` > `Has DC system`

DanielMcInnes, 2024-12-29:
Hi @philipptrenz , currently in gui-v2 we have a 'Services' page in Settings. It contains the 'tailscale' page, CANbus settings, and a switch for CANbus over tcp/ip. I assume that should go in the new 'Connectivity' page?

philipptrenz, 2024-12-31:
Hi @DanielMcInnes,

yes, I'd say Tailscale is best placed within "Connectivity". Also the CAN ports are meant to be placed there, see hierarchy tree in Figma. Regarding the "CANbus over tcp/ip" toggle, I guess this is a per CAN port setting and can be placed under the related CAN interface?

DanielMcInnes, 2024-12-31:
> Hi @DanielMcInnes,
> 
> yes, I'd say Tailscale is best placed within "Connectivity". Also the CAN ports are meant to be placed there, see hierarchy tree in Figma. Regarding the "CANbus over tcp/ip" toggle, I guess this is a per CAN port setting and can be placed under the related CAN interface?

Thanks @philipptrenz , sounds good

DanielMcInnes, 2025-01-02:
Hi @philipptrenz , what is meant to go in the 'Display Units' page? it seems like a duplicate of 'Data Units'. Also, in 'Data Units', the new design has 'speed' and 'gps format'. I think we support multiple gps receivers, and allow configuring speed & gps format for each gps receiver, so there's not a single system-wide setting for speed & gps format, you have to navigate to a specific gps receiver and configure it there.

closed 2025-01-13 by DanielMcInnes

mr-manuel, 2025-01-13:
1. If none of the large features are available/visible, the note should not be displayed.

[image]

2. The Venus OS large features are not visible at all, even not as `Superuser`.

3. Should Debug & Develop always be displayed? Before it was shown only when `Superuser` access level was enabled.

mr-manuel, 2025-01-15:
@DanielMcInnes should I open a new issue for that?

DanielMcInnes, 2025-01-20:
> [@DanielMcInnes](https://github.com/DanielMcInnes) should I open a new issue for that?

Thanks @mr-manuel , I created a PR for this

mr-manuel, 2025-01-21:
Looks good now

commit c619e905 (2024-12-18): Redesign settings layout

Fixes #1232
commit 09153025 (2024-12-23): Redesign settings layout

Fixes #1232
commit 36d0b836 (2025-01-01): Redesign settings layout

Fixes #1232
commit 3e122fe7 (2025-01-01): Redesign settings layout

Fixes #1232